esphome/feature-requests #1153: Add support for the Waveshare 3.7 inch display (with partial update)

bjobo, 2021-03-04:
**Describe the problem you have/What new integration you would like**
Add the Waveshare 3.7inch e-Paper e-Ink Display to the Waveshare integration.

**Please describe your use case for this integration and alternatives you've tried:**
The Waveshare 3.7inch e-Paper e-Ink Display is (currently) the biggest Waveshare with partial refresh (and SPI). So it would be great to add it as supported device. Also it seems (at the first glance) pretty straight forward to integration. The specs seem similar to the 2.9inch (both have partial update) or the 4.2inch.
I found this repo where both displays are working: https://github.com/ZinggJM/GxEPD _*)_ and compared the code for the 2.9inch and the 3.7inch display: Both are very similar but with differences in the 'wake up' and the 'full update' method. If these differences are meaningfull or 'bad coding' (due to copy and paste from the examples) I don't know.

_*) The Repo is not primarly designated for Waveshare displays but for 'Good Display'. However, they seem to use the same panels and boards)._

**Additional context**
Device specs: https://www.waveshare.com/3.7inch-e-Paper.htm

_Side note: The 3.9inch as well as the 4.2inch have a 4 colour grey scale. As far as I understood, this is not supported by the current ESPHome implementation_

maxmeyer, 2021-05-03:
@bjobo But the b/w mode is supported for those displays? Even the red ones? I'm a bit lost what I can expect by the current implementation. Thanks in advance.

tempico, 2021-10-30:
Agree with @bjobo 
I would love to see that display in the list of supported ones and I am ready to donate $150 if it can be implemented and merged into main branch by the end of 2021.

SagiosNL, 2023-07-01:
+1 for me as well.

czepter, 2024-11-13:
any development here on this?