esphome/feature-requests #1155: SCD30: Add Forced Re-Calibration method

Lukaboo, 2021-03-07:
**Describe the problem you have/What new integration you would like**
At the moment the SCD30 integration only supports automatic calibration. For accurate readings the sensor has to be exposed to clean air daily. Otherwise it will see the "not so fresh air" in the room as a new fresh air reference. Based on the adafruit library, there is a method for forced recalibration ( similar to mhz19.calibrate_zero Action of the mhz19 co2 sensor).

**Please describe your use case for this integration and alternatives you've tried:**
With this method automatic calibration does not leed to false readings. Just expose the SCD-30 to fresh outside air and use a value of 400 ppm.

**Additional context**
Here is a better description of this feature: https://learn.adafruit.com/adafruit-scd30/field-calibration

dracula385, 2021-05-04:
+1 for this feature request.. I cannot confidently use SCD-30 without FRC: below are additional documents regarding FRC

**Page 14**
https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/9.5_CO2/Sensirion_CO2_Sensors_SCD30_Interface_Description.pdf

**Page 3**
https://www.mouser.com/pdfdocs/CD_AN_SCD30_Low_Power_Mode_D2.pdf

marcofl, 2021-12-21:
From my own SCD30 mqtt project, I've a button that I can press to calibrate to 400ppm. I can easily take the esp32 with sensor outside (e.g. battery power it, smart phone, etc.), leave it for 5min and press the button. Calibration is saved to sensor NVRAM, nothing more to do on the ESP side.

for ESPHome / Home Assistant, I think we could also just have a switch in Home Assistant to trigger the calibration.

My relevant code pieces, if that helps:

```
#include <Adafruit_SCD30.h>
#include <GFButton.h>

Adafruit_SCD30  scd30;
GFButton recalibrateButton(27); // D4 on firebeetle 2 ESP32

void recalibrateSensor(int value) {
  if (!scd30.forceRecalibrationWithReference(value)){
    Serial.println("Failed to force recalibration");
  } else {
      Serial.println("Recalibrated to" + (String)value);
  }
}

void setup() {
  if (!scd30.begin()) {
    Serial.println("Failed to find SCD30 chip");
    while (1) { delay(10); }
  }
}

void loop() {
  if (recalibrateButton.wasPressed()) {
    Serial.println("Button pressed, recalibrating to" + (String)recalibrationValue + " ppm");
    recalibrateSensor(400);
  }
}

```

wtip, 2022-01-23:
Just FYI there's currently a pull request for this feature if you want to watch it https://github.com/esphome/esphome/pull/1237