esphome/feature-requests #1157: Getting response from http_request

qqgg231, 2021-03-08:
**Describe the problem you have/What new integration you would like**

I have button and LED.
When button is clicked, http_request is made.

I want to have feedback if this action is successful. When this request is successful, the LED should turn on for 1 sec.

So I need to be able to write (pseudocode):

```
If ( request_response == "finished successfully")
then:  ...

```
or version with checking just http status code:

```
If ( request_status_code == 200)
then:  ...
```

Any of those versions would satisfy me needs. I just need to have feedback to instatly know that action was successful.

AFAIK it is not possible to be done currently.

**Please describe your use case for this integration and alternatives you've tried:**

Included above.

**Additional context**

andreashergert1984, 2021-03-29:
HI, there is already something like that im implemented (just not documented). 
there is an "on_response" option where the response code gets in as a parameter.

glmnet, 2021-03-29:
here are the docs, it's beta and dev https://beta.esphome.io/components/http_request.html#on-response-trigger

f403, 2021-12-14:
Obtaining a response body would be also nice

nagyrobi, 2022-02-23:
See https://github.com/esphome/esphome-docs/pull/1914

closed 2022-07-01 by nagyrobi