esphome/feature-requests #1158: Publish device_class via MQTT Autodiscovery

mschae, 2021-03-08:
**Describe the problem you have/What new integration you would like**

- Home Assistant uses matching icons for different `device_class`es of the same entity type (in my example the `garage` `device_class` of the cover entity type.
- Publishing the `device_class` is possible via the MQTT autodiscovery message. 
- Since we can already configure it, we should publish it. 

**Please describe your use case for this integration and alternatives you've tried:**

See above: Customizing the appearance of entities in home assistant. 
A workaround is configuring the device class via custom configs as suggested in #553. 

**Additional context**

definitio, 2021-07-26:
Added in 1.20.0

krconv, 2021-07-29:
Love the fix @definitio; looks like we can do the same thing for covers, which fixes this for garage doors (https://github.com/esphome/esphome/pull/2092)

closed 2021-07-29 by OttoWinter