esphome/feature-requests #1163: single point thermostats should work after power up without needing to be changed into a new mode

larsonmpdx, 2021-03-14:
**Describe the problem you have/What new integration you would like**

see https://esphome.io/components/climate/thermostat.html
Single point thermostats (those with either **default_target_temperature_low** or **default_target_temperature_high** set but not both) should default to heat operation or cool operation so that they'll function correctly after a power loss, or they should function correctly under "heat/cool" operation.

**Please describe your use case for this integration and alternatives you've tried:**

When single point (cooling only or heating only) thermostats turn on they default to "heat/cool" operation and in that mode they won't activate.  They do work correctly after being switched to "heat" or "cool" operation.  I was able to work around this by setting both a **default_target_temperature_low** and **default_target_temperature_high** even though I only want cooling operation - I made the heating temperature arbitrarily low, and now in the default "heat/cool" mode it does work.  I assume using bang-bang would be an option as well.

**Additional context**

larsonmpdx, 2021-03-15:
the problem looks like it always defaults to auto (heat/cool) mode here:
https://github.com/esphome/esphome/blob/dev/esphome/components/thermostat/thermostat_climate.cpp#L24

then in this switch statement, when it doesn't have two point control, there's no case statement for auto mode:
https://github.com/esphome/esphome/blob/dev/esphome/components/thermostat/thermostat_climate.cpp#L197

a fix could be to look at this->supports_two_points_, this->supports_cool_ and this->supports_heat_ in the setup function and make the default mode heat or cool as needed

can someone who understands the code more tell me if that might work? I can make a pr if so

jarrettv, 2021-03-19:
@kbx81

kbx81, 2021-03-20:
This is interesting -- it _should_ restore whatever mode it was in prior to power loss. Mine do...although admittedly I don't have a single-point config set up right now. Regardless, I would expect the restore behavior to be the same.

It might be possible to fix this by simply removing https://github.com/esphome/esphome/blob/dev/esphome/components/thermostat/thermostat_climate.cpp#L24 
...and adding something to validate the mode after restore (to ensure it didn't restore garbage) and setting to an appropriate mode if it ends up in an invalid one.

kbx81, 2021-03-20:
There are a couple other issues open with `thermostat` right now, I was planning to work on these fixes after some of the other in-progress climate changes get merged in. If this is important to you, feel free to create a PR and I'll review it. I don't power-cycle or reset my thermostats very often so it's one of those issues that has slipped through the cracks. 😇

larsonmpdx, 2021-04-15:
thanks I'll do this fix it should be only a few lines

I think restoring correctly depends on the user's choice for esp8266_restore_from_flash

closed 2021-07-21 by jesserockz