esphome/feature-requests #1166: ESPHome based MQTT gateway for nodes using non-wifi communication

Laczen, 2021-03-19:
**Describe the problem you have/What new integration you would like**

At the moment ESPHome communication with HA servers is limited to wifi only systems. There are many non-wifi communication methods like espnow, lora, ble that are not or have limited support by ESPHome.

A general approach that allows for easy integration of nodes that are using a different communication method could be made by creating a ESPHome based (transparent) MQTT gateway that transfers the mqtt messages over espnow, lora, ble, ... and then at the node (that could also be ESPHome based) use the messages as a mqtt client.

The message transformation could be done as packing the mqtt message into a json package containing `{ n: node, t: topic, p: payload}`.

The idea would translate to:
`HASS <--tcp/ip--> MQTT-broker <--tcp/ip--> MQTT-gateway (ESPHome) <--espnow,lora,ble--> MQTT-client (ESPHome, node)`

The MQTT-gateway would be used only as a translater of packages and could do a broadcast of the packages, the client then decides if this is something it needs to react upon or to ignore.  

**Please describe your use case for this integration and alternatives you've tried:**

***use case:***
My use case consists of many nodes (light switches) that I do not want to connect to my wifi network (because of security, wifi load, messages are scarce, ...), I want these nodes to communicate over espnow and to report their state to HASS. HASS commanding the nodes would be a scarce thing.

***alternatives:***
A different approach could be to use the same method for the ESPHome API, but this would have several disadvantages:
a. In the API the nodes are servers, that does not work well with nodes that are sleeping most of the time,
b. The API is a more or less HASS only solution, while MQTT is very well known and widely used.

**Additional context**

Laczen, 2021-03-19:
@SenexCrenshaw, fyi

749, 2022-03-25:
This seems like the same direction as #275 and #124