esphome/feature-requests #1167: New 2.13 inch epaper model type: b74 (GDEM0213B74)

matikij, 2021-03-19:
**Describe the problem you have/What new integration you would like**

There is a new model basting 4-level grayscale https://www.good-display.com/product/375.html
there are some code samples here:

https://www.good-display.com/companyfile/555.html
https://www.good-display.com/companyfile/556.html

Nothing in GxEPD2 library repo yet https://github.com/ZinggJM/GxEPD2/tree/master/src/epd .

Would be cool if this would be supported at some point.

Didn't delete any bolierplate text from this issue.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nikito7, 2021-03-19:
https://github.com/lewisxhe/GxEPD/tree/master/src/GxGDEM0213B74

nikito7, 2021-05-06:
It's a new type. No lut.

It will not be easy...

Advanced coder needed

osresearch, 2021-06-09:
[image]

Preliminary support for using the b74 in monochrome: https://github.com/esphome/esphome/pull/1869

nikito7, 2021-06-09:
Board is new too: 2.3.1

OttoWinter, 2021-08-10:
Closed via https://github.com/esphome/esphome/pull/1869

closed 2021-08-10 by OttoWinter