esphome/feature-requests #1169: Support for SJWS01LM MI Flood Detector

varsanyim, 2021-03-20:
**Describe the problem you have/What new integration you would like**

Please add support for the SJWS01LM Xiaomi MI Flood Detector

**Please describe your use case for this integration and alternatives you've tried:**

I did not try, but afaik the ble_monitoring Home Assistant custom component supports it.

**Additional context**

If I need to test anything, just give me a ping ;)

dentra, 2021-08-17:
You can take it here: https://github.com/dentra/esphome-components/tree/master/components/miot_sjws01lm

It implementation is quite different from ESPHome Xiaomi platform, but by mi specs, more clear and without magic and now under heavy development to conform need of other devices.

mamartins, 2022-04-30:
@dentra is it possible to use it without a bind key? I don't have the gateway only the ESPHome

dentra, 2022-05-01:
@mamartins, gateway is not needed. You only need to connect the device to MiHome app once. Then use my integration to automatically obtain bindkey.

mamartins, 2022-05-02:
The app didn't allow to connect to this device without the hub, I as able to get it via https://atc1441.github.io/TelinkFlasher.html

No battery or voltage info is received after 2 days but flood is working after a few rebuilds

dentra, 2022-05-05:
@mamartins do you have any logs associated with this device?

mamartins, 2022-06-03:
I don't but happy to get some, how should I do?

closed 2022-07-05 by nagyrobi