esphome/feature-requests #1171: Add

keesma, 2021-03-21:
**Describe the problem you have/What new integration you would like**

I am using the esp32_ble_tracker to collect data from multiple ble sensors. The sensors have multiple values. I would like to improve the layout of the mqtt messages by adding an extra layer in the mqtt label hierarchy using the sensor name.

Currently it look like this:
/node-name/sensor/atc_temperature_bathroom
/node-name/sensor/atc_humidity_bathroom
/node-name/sensor/atc_battry-voltage_bathroom
/node-name/sensor/atc_battry-level_bathroom
/node-name/sensor/atc_temperature_sleepingroom
/node-name/sensor/atc_humidity_sleepingroom
/node-name/sensor/atc_battry-voltage_sleepingroom
/node-name/sensor/atc_battry-level_sleepingroom

The desired structure is:
/node-name/sensor/bathroom/atc_temperature
/node-name/sensor/bathroom/atc_humidity
/node-name/sensor/bathroom/atc_battry-voltage
/node-name/sensor/bathroom/atc_battry-level
/node-name/sensor/sleepingroom/atc_temperature_
/node-name/sensor/sleepingroom/atc_humidity
/node-name/sensor/sleepingroom/atc_battry-voltage
/node-name/sensor/sleepingroom/atc_battry-level

**Please describe your use case for this integration and alternatives you've tried:**

The mqtt label hierarchy is more clear in this way.

**Additional context**

Laczen, 2021-03-23:
@keesma, I think this could be achieved by adding a location property to all components. This would benefit a lot of applications, not only mqtt.

keesma, 2021-03-23:
@Laczen: nice addition! In my case it is indeed the location. if a sensor can be named, you have the flexibility to either use the location or an other criterion for clustering sensor data.

Laczen, 2021-03-23:
@keesma, I have also created a discussion item under esphome/esphome.

keesma, 2021-03-26:
I define my sensor now like this (I have a lot of temperature/humidity sensors in my house so the list is much longer):

sensor:
  - platform: uptime
    name: "${esphome_name} Uptime Sensor"
    
  - platform: wifi_signal
    name: "${esphome_name} WiFi Signal"
    update_interval: 60s

  - platform: ble_rssi
    mac_address: xx:xx:xx:xx:xx:xx
    name: "ATC bathroom"
 
  - platform: atc_mithermometer
    mac_address: xx:xx:xx:xx:xx:xx
    temperature:
      name: "ATC Temperature - bathroom"
    humidity:
      name: "ATC Humidity - bathroom"
    battery_level:
      name: "ATC Batterylevel - bathroom"
    battery_voltage:
      name: "ATC Batteryvoltage - bathroom"

  - platform: ble_rssi
    mac_address: A4:C1:38:58:5C:95
    name: "ATC bedroom"

  - platform: atc_mithermometer
    mac_address: xx:xx:xx:xx:xx:xx
    temperature:
      name: "ATC Temperature  - bedroom"
    humidity:
      name: "ATC Humidity  - bedroom"
    battery_level:
      name: "ATC Batterylevel  - bedroom"
    battery_voltage:
      name: "ATC Batteryvoltage  - bedroom"