esphome/feature-requests #1172: ble_presence: Add option to wait XX seconds before changing state

chemelli74, 2021-03-21:
**Describe the problem you have/What new integration you would like**

New option for `ble_presence`: _change_state_after_

**Please describe your use case for this integration and alternatives you've tried:**

Would be nice to have the possibility to specify a time frame needed to consider a change state for the device as some devices sometimes loose connection for just 1/2 sec.
Been able to avoid a state change in such situations would make HA state machine and HA database happier.

**Additional context**

nikito7, 2021-03-22:
```yaml
      filters:
      - delayed_on: 2s
      - delayed_off: 2s
```

chemelli74, 2021-03-22:
Thx a lot for the tip.

Simone

chemelli74, 2021-03-22:
Unfortunately it doens't work here:

chemelli74, 2021-03-23:
Sorry just realized this is a FR.
Creating a issue.

Simone

closed 2021-03-23 by chemelli74