esphome/feature-requests #1173: Add Heltec Wifi Kit 32 V2

Noxmiles, 2021-03-22:
Since 2019 there is Version 2 of the Heltec Wifi Kit 32. This new Version has some differences, which should be updated in ESPHome:
- 4MB (32M-bits) FLASH upgrade to 8MB (64M-bits) FLASH
- Different Pinout: See https://heltec-automation-docs.readthedocs.io/en/latest/esp32+arduino/wifi_kit_32/hardware_update_log.html for the differences

Especially with 8MB the limit for the firmware size is doubled.

There is also V2.1, but the chances does not affect ESPHome

richardklingler, 2021-05-18:
The changes in the pinout doesn't matter to ESPHome as you specify the GPIOs to use...

Also for ESP32 flash size doesn't matter...no ESP32 board has a flash size defined, only ESP8266 boards...

nagyrobi, 2022-07-07:
Closing as the answer above is clear.

closed 2022-07-07 by nagyrobi