esphome/feature-requests #1174: US wire based thermostat component

jarrettv, 2021-03-24:
**Describe the problem you have/What new integration you would like**
The current configuration for `thermostat` is very flexible but also complex. I'd like to see something that works more in line with a typical off the shelf thermostat automatically. For example, for my heat pump with 2 stage (emergency heat), the config would be:
```climate:
  - platform: thermostat_us
    id: ${node_name}_thermostat
    name: ${node_friendly_name}
    visual:
      min_temperature: 55°F
      max_temperature: 92°F
      temperature_step: 1
    sensor: ${node_name}_bme280_temp
    hysteresis: 0.555555555555556
    emergency_heat_mode: eco
    default_target_temperature_low: 67°F
    default_target_temperature_high: 72°F
    wires:
      - fan: ${node_name}_triac_green_wire
      - compressor: ${node_name}_triac_yellow_wire
      - reversing_valve: ${node_name}_triac_orange_wire
      - emergency_heat: ${node_name}_triac_white_wire```

**Please describe your use case for this integration and alternatives you've tried:**
I've tried the existing `thermostat` and yes it will satisfy my use case, however it is more complicated than expected.

**Additional context**
By default it should support fan: ON/AUTO, mode: OFF/COOL/HEAT/EM HEAT/AUTO and should have all the logic for how a typical thermostat would work. For example, I think most thermostats call for emergency heat if you have +2 degree temperature delta. You can override default behavior with a different `emergency_heat_mode`.

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE FEATURE REQUEST WILL BE CLOSED AUTOMATICALLY. -->

jarrettv, 2021-03-25:
I'd also like to see some of these compressor and other settings included and defaulted in the thermostat logic:

https://support.ecobee.com/hc/en-us/articles/360002020552-Threshold-settings-for-ecobee-thermostats

jarrettv, 2021-03-25:
Potential rule additions, + likely need to include, - too advanced for now, ? not sure

+Auto Heat/Cool Minimum Delta, 5F 2.8C
+Heat Dissipation Time 15s?
+Cool Dissipation Time 15s?
+Heat Diff Temp = 0.5F 0.3C
+Cool Diff Temp = 0.5F 0.3C
+Heat Min On Time = 300s
-AC Overcool Max = Humidity
-Temperature Correction
-Humidity Correction
+Compressor Minimum Cycle Off Time 300s
+Compressor Minimum On Time 300s
-Compressor Minimum Outdoor Temperature: depends on region and equipment 22F or -2C
-Compressor Stage 2 Temperature Delta
-Compressor Reverse Staging
-Compressor Stage 1 Max Runtime

-Heat Stage 2 Temperature Delta
-Heat Stage 1 Max Runtime

+Aux Heat Max Outdoor Temperature = default none?
+Aux Heat Minimum On Time = yuk 300s
?Compressor to Aux Temperature Delta = auto
?Compressor to Aux Runtime = auto
-Aux Stage 2 Temperature Delta
-Aux Reverse Staging
-Aux Stage 1 Max Runtime

kbx81, 2021-03-28:
I've been kicking this idea around a lot in my head and, rather than making a separate component (which will likely bring about more confusion), I think it would make more sense to extend the functionality of the existing `thermostat` component. I had an idea that I think would work well...

We could add a `wires` (or honestly I think `gpios` would make more sense as it relates to ESPHome) section to the existing component's configuration. So, for example, you could have:

```yaml
# Example dual-point configuration entry
climate:
  - platform: thermostat
    name: "Thermostat Climate Controller"
    sensor: some_temperature_sensor
    default_target_temperature_low: 20 °C
    default_target_temperature_high: 22 °C
    cool_action:
      - script.execute: some_cool_script
    heat_action:
      - script.execute: some_heat_script
    idle_action:
      - script.execute: some_off_script
    gpios:
      - fan: triac_green_wire
      - compressor: triac_yellow_wire
      - reversing_valve: triac_orange_wire
      - aux_heat: triac_white_wire
```

The other options you're talking about adding would be pretty simple to bring into the configuration, too. They'd just end up as additional configuration options along side of those that already exist.

jarrettv, 2021-03-29:
I like it. I've spent some quality time with the thermostat component and it makes sense to build it into existing one. I like `gpios` or maybe even `switches` since that is component being hooked up (similar to `sensor` for temp).

I think the first couple safety threshold configs should be:
- Compressor Minimum Cycle Off Time 300s
- Compressor Minimum On Time 300s