esphome/feature-requests #1175: Adding BLE support for Eqiva Ble Smart Lock

nerdosity, 2021-03-25:
**Describe the problem you have/What new integration you would like**
Integration for EQ3 HomeMatic smart lock
https://www.eq-3.com/products/homematic/detail/bluetooth-smart-lock.html

**Please describe your use case for this integration and alternatives you've tried:**
This is already working, it seems, but it's not ESPHome :) 
https://github.com/tc-maxx/esp32-keyble/blob/master/src/main.cpp

**Additional context**
This is a good, cheap lock, impossible to integrate in Home assistant.
It would be much more useful with a proper integration

h0jeZvgoxFepBQ2C, 2022-09-19:
With the new bluetooth proxy support, would it be possible to just forward the equiva bluetooth data to home assistant via proxy?

digaus, 2023-10-24:
There is also this repo: https://github.com/mccoy88f/esp32-keyble-ha

Would be nice if this could be added to esphome :)

Then it would be easy to integrate with a grow fingerprint

digaus, 2023-11-22:
Since I am designed my own board now to be specifically used with a grow fingerprint and an Eqiva BLE lock I am in the process of porting this to esphome:

Be aware I am not a C++ dev and basically just programming it by trial and error. However I am currently already able to connect to the Eqiva Lock and exchange the nonce (get the remote security counter)

So fingers crossed :)

[image]

digaus, 2023-11-28:
> Don't forget battery status...

Only can get info if battery is low :)

[image]

digaus, 2023-11-29:
[image]

HA services working fine :)

digaus, 2023-11-30:
> @digaus do you have your working state somewhere checked in? I'd be interested to test it.

If you really want to: https://github.com/digaus/esphome-components-eqiva

With service pair you can create a new user_key and user_id

Currently it will be logged only

Strakomania, 2023-12-12:
Hey guys im very interested in testing the esphome configuration with my equiva Lock.
But i am a newby in esphome and dont know how to set this up. 
Is it just copy and paste? 
Maybe you have a small instruction for me ? :)

scrivoluce, 2023-12-12:
Awesome,thanks for you work! I've just flashed esphome to my first esp32. I'll git it a try and feedback.

Strakomania, 2023-12-13:
@scrivoluce i also flashed esphome to my esp32 but dont know whats the next step. Can you tell me how you get it to work? 

I would be very thankful

ginko64, 2023-12-13:
I also have the Eqiva lock and am very interested in this project. What I don't understand is how to pass the access credentials contained in the original QRcode to EspHome

digaus, 2023-12-13:
@scrivoluce  and @Strakomania 

https://github.com/digaus/esphome-components-eqiva/blob/main/README.md#initial-pairing-after-you-have-done-the-previous-steps

Updated the readme

digaus, 2023-12-13:
> @digaus you wrote in the new readme: "need to scan QR code of the card and copy the result" how can I do this?
> 
> thanks for your patience :)

Smartphone? :)

ginko64, 2023-12-14:
> > @digaus you wrote in the new readme: "need to scan QR code of the card and copy the result" how can I do this?
> > thanks for your patience :)
> 
> Smartphone? :)

It was easy :) 
thank you

Strakomania, 2023-12-14:
@ginko64 

Can you show me your yaml Code you get to Work? 
When i copy the Parts from the Readme the Installation will fail

digaus, 2023-12-14:
Check the example yaml

ginko64, 2023-12-14:
> @ginko64
> 
> Can you show me your yaml Code you get to Work? When i copy the Parts from the Readme the Installation will fail

I haven't done it yet, I'm waiting for the esp32 from Amazon. 
I put yaml file (from readme) in  my esphome folder in Home Assistant but I have several errors. For now I try to figure out

Strakomania, 2023-12-14:
oh okay.
This is also my problem i put the code snippets together and compare it with the "example.yaml" put i get just errors.

digaus, 2023-12-14:
Just go to esphome menu and click on add device. Then paste the example yaml

digaus, 2023-12-14:
What errors do you specifically get?

Strakomania, 2023-12-14:
> Just go to esphome menu and click on add device. Then paste the example yaml

okay let me triy it this way at first.

Strakomania, 2023-12-14:
> What errors do you specifically get?

INFO ESPHome 2023.11.6
INFO Reading configuration /config/esphome/equiva-lock.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-eqiva-lock (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/eqiva_key_ble/eqiva_key_ble.o
src/esphome/components/eqiva_key_ble/eqiva_key_ble.cpp:358:314: warning: multi-character character constant [-Wmultichar]
       ESP_LOGI(TAG, "Waiting for connection... (sendingNonce: %s) (remoteSessionNonce: %s) (remoteSessionNonce: %s) (clientStateEstablished: %s)", sendingNonce ? 'true' : 'false', clientState.remote_session_nonce.length() > 0 ? 'true' : 'false', this->state() == espbt::ClientState::ESTABLISHED ? 'true' : 'false');
                                                                                                                                                                                                                                                                                                                          ^
src/esphome/components/eqiva_key_ble/eqiva_key_ble.cpp:358:314: warning: character constant too long for its type
src/esphome/components/eqiva_key_ble/eqiva_key_ble.cpp:358:314: warning: multi-character character constant [-Wmultichar]
src/esphome/components/eqiva_key_ble/eqiva_key_ble.cpp:358:314: warning: character constant too long for its type
src/esphome/components/eqiva_key_ble/eqiva_key_ble.cpp:358:314: warning: multi-character character constant [-Wmultichar]
src/esphome/components/eqiva_key_ble/eqiva_key_ble.cpp:358:314: warning: character constant too long for its type
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/text_sensor/text_sensor.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/time/automation.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/time/real_time_clock.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/web_server/list_entities.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/web_server/web_server.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/web_server_base/web_server_base.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/web_server_idf/web_server_idf.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/wifi/wifi_component.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/wifi/wifi_component_esp32_arduino.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/wifi/wifi_component_esp8266.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/wifi/wifi_component_esp_idf.o
In file included from src/esphome/components/eqiva_key_ble/eqiva_key_ble.cpp:2:
src/esphome/components/eqiva_key_ble/eqiva_key_ble.cpp: In member function 'bool esphome::eqiva_key_ble::EqivaKeyBle::sendMessage(eQ3Message::Message*, bool)':
src/esphome/components/eqiva_key_ble/eqiva_key_ble.cpp:358:21: error: format '%s' expects argument of type 'char*', but argument 5 has type 'int' [-Werror=format=]
       ESP_LOGI(TAG, "Waiting for connection... (sendingNonce: %s) (remoteSessionNonce: %s) (remoteSessionNonce: %s) (clientStateEstablished: %s)", sendingNonce ? 'true' : 'false', clientState.remote_session_nonce.length() > 0 ? 'true' : 'false', this->state() == espbt::ClientState::ESTABLISHED ? 'true' : 'false');
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:151:28: note: in expansion of macro 'esph_log_i'
 #define ESP_LOGI(tag, ...) esph_log_i(tag, __VA_ARGS__)
                            ^~~~~~~~~~
src/esphome/components/eqiva_key_ble/eqiva_key_ble.cpp:358:7: note: in expansion of macro 'ESP_LOGI'
       ESP_LOGI(TAG, "Waiting for connection... (sendingNonce: %s) (remoteSessionNonce: %s) (remoteSessionNonce: %s) (clientStateEstablished: %s)", sendingNonce ? 'true' : 'false', clientState.remote_session_nonce.length() > 0 ? 'true' : 'false', this->state() == espbt::ClientState::ESTABLISHED ? 'true' : 'false');
       ^~~~~~~~
src/esphome/components/eqiva_key_ble/eqiva_key_ble.cpp:358:21: error: format '%s' expects argument of type 'char*', but argument 6 has type 'int' [-Werror=format=]
       ESP_LOGI(TAG, "Waiting for connection... (sendingNonce: %s) (remoteSessionNonce: %s) (remoteSessionNonce: %s) (clientStateEstablished: %s)", sendingNonce ? 'true' : 'false', clientState.remote_session_nonce.length() > 0 ? 'true' : 'false', this->state() == espbt::ClientState::ESTABLISHED ? 'true' : 'false');
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
[... cut ...]
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/components/wifi/wifi_component_pico_w.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/core/application.o
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/core/color.o
cc1plus: some warnings being treated as errors
Compiling .pioenvs/esphome-eqiva-lock/src/esphome/core/component.o
*** [.pioenvs/esphome-eqiva-lock/src/esphome/components/eqiva_key_ble/eqiva_key_ble.o] Error 1
========================== [FAILED] Took 4.68 seconds ==========================

digaus, 2023-12-14:
Add `refresh: 0s` below the GitHub url

Strakomania, 2023-12-14:
there already was added 0s.

i had to copy and paste the raw code wich i get when i click on the copy button?

digaus, 2023-12-14:
Remove the # on that line, otherwise it's only a comment...

Strakomania, 2023-12-14:
oh yeah that was stupid , sorry :D

Strakomania, 2023-12-14:
so i succeed to flash the Firmware :)

at the end i got this Log:

========================= [SUCCESS] Took 9.15 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of esphome-eqiva-lock.local
ERROR Error resolving IP address of esphome-eqiva-lock.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname

What can i doo?

digaus, 2023-12-14:
Verify your WiFi credentials? Connect ESP via USB and check log output on https://web.esphome.io ?

Strakomania, 2023-12-14:
Now i got it!

Addeed a static IP and i had to add the OTA Password. Now it succsessfully flashed tot the esp32
I got 19 entities for my Equiva lock Esp :) 

So all in all i flashed the Esp32 over https://web.esphome.io/ an after that i copied the raw code from github.
Then i had to change the wifi settings an ad the OTA Password and ad the Passkey i get after flashing. 

Then install it. 

thank yout for the help :))

Strakomania, 2023-12-14:
sorry but i got the next issue 

I added my lock and it looks pretty good but i can´t controll ost and i dont see the status.

the Lock Ble State switches every second the state 

[image]

[image]

[image]

Strakomania, 2023-12-14:
The BLE State shows Established but everytime i click "aufschließen" the state changes to connect and idle

digaus, 2023-12-14:
> The BLE State shows Established but everytime i click "aufschließen" the state changes to connect and idle

Have you done the pairing process with the ESP?

The lock needs to beep and then you get user_id and user_key.

Also after pair you should remove the card_key

Maybe it is also good to not spam this issue but better create an issue on my repo :)

Strakomania, 2023-12-14:
I think i did something wrong while the pairing. 
I tried it again step by step with your instruction and i it work. :)

Sorry for spaming im a newbie on gitub so i have to figure out how to behave. Next time will i open a new issue . :)

ginko64, 2023-12-14:
@Strakomania 
could you contact me privately? so help me with the lock and let's avoid spam on this page
mario.r@me.com

thanks

ginko64, 2023-12-14:
@Strakomania 
can you help me with the configuration in private? So we do not spam the topic
mario.r@me.com
thanks

scrivoluce, 2023-12-17:
Kudos @digaus ! 
I just got the time day to play with it! THIS IS AWESOME!
It worked like charm!!!
So gratefull for you work!

sobrarbe, 2024-04-06:
Hello, @digaus 
 I am really very excited about this project. I have been wanting to put my eqavi lock on home assitnat for a while. but I have a problem. at the time I put the mac in the ESPHome Eqiva Lock. It doesn't connect to me and it keeps finding the mac and it doesn't come out connected. Can you help me to make it work? It is very important to me, thank you for your great work.
[image]