esphome/feature-requests #1176: Add parameter to LEDC to set what should be 1% value, so lights actually turn on at 1%, and not at a higher value

Timmiej93, 2021-03-26:
**Describe the problem you have/What new integration you would like**

I'm using LEDC to dim my LED strips. These LED strips only come on at a certain value though, say 10%. This means that the usable range for dimming is [10-100%]. I would like to be able to add a minimum value for dimming, so that the 10-100% range gets scaled to be 0-100%, i.e. what was previously the 10% dimming value now becomes the 1% dimming value, and everything between that and 100% is scaled accordingly.

Proposal:
```output:
  - platform: ledc
    ...
    min_brightness: x
    ```
min_brightness would be a value that can specify which value to use as the first value in the used UI. Something like 10% or 0.10.

**Please describe your use case for this integration and alternatives you've tried:**
<!-- A clear and concise description of what the problem is. -->
I tried light.core's min_power setting, but this makes it impossible to turn the lights off, since min_power == off. When I set the min_power value to 0.5 (50%) for testing, it was impossible to turn off the lights. Although theoretically not a problem if I set the min_power value to a value just below where the LEDs come on, this means that there's always power going through the LEDs, and I'd rather have off really be off, and not "on, but not enough to see it". 

Current situation:
 - Set output to anything from 1% to 9%: nothing happens
 - Set output to 10%, lights come on

Desired situation:
 - Set output to 1%, lights come on

**Additional context**
<!-- Add any other context about the feature request here. -->

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE FEATURE REQUEST WILL BE CLOSED AUTOMATICALLY. -->

DavidRajchman, 2021-04-05:
would love to see this

prclm, 2021-06-29:
i have the same problem an like to have this solution 👍

pat1300, 2021-06-30:
I also vote for this enhancement

prclm, 2021-07-06:
@Timmiej93 @DavidRajchman  @pat1300 

here is a workaround:
```yaml
# Output (with conversion)
output:
  - platform: template
    id: output_conversion
    type: float
    write_action:
      - lambda: 'id(gpio_5).set_level(state == 0 ? 0 : state * 0.977 + 0.023);'
  - platform: ledc
    id: gpio_5
    pin: 5
```

thanks to @ssieb from the discord channel!

Timmiej93, 2021-07-06:
@prclm That's brilliant, I love that you can do that. Also, thank you very much for sharing this.
I had to do quite a bit of research before I fully understood how the template works, so I'll describe what I think is true here. Please correct me if I'm wrong. I'll also add an example YAML file to see if I understood the implementation correctly.

```
output:
  - platform: template
    id: light1_template
    type: float
    write_action:
      - lambda: 'id(light1).set_level(state == 0 ? 0 : state * ... + ...);'
  - platform: ledc
    id: light1
    pin: 1

light:
  - platform: monochromatic
    name: Light1
    output: light1_template
```

The `light` section creates a dimmable light element in HA. When the value of this element is changed in HA, the value is sent to the output with id `light1_template`. This triggers the template, which calls the line of code, which either sets the light output to 0, or to the correctly scaled new value.

Something I'm wondering about though: would it be better to use `output.turn_off` instead of `output.set_level`? I don't know what `.set_level(0)` does. It would make sense for that to be the same as `.turn_off`, but is that factually true, or do we only think it's true? I get that the `.set_level` method makes for a shorter and better looking oneliner, but it's not like we're looking at these files often, so if `.turn_off` is better, I'd personally use that. Question being: is there a difference?

About the formula, since it may be a bit obscure for those who aren't programmers, I'll clarify it a bit for those that have trouble understanding it. If I misinterpreted it, please feel free to correct me.

```
id([output_id]).set_level(state == 0 ? 0 : state * [value1] + [value2])
```

- `output_id`: This is the ID of the LEDC component. You assign this ID yourself in your YAML file.
- `value2`: This value denotes the **un**-usable range of your light. If (with the normal setup) your light turns on at 20%, this value would be `0.2`.
- `value1`: This value denotes the usable range of your light. To calculate this value, simply calculate `1 - value2`. With the `0.2` value determined in this example, `value1` would be `0.8`.

Again, thank you for sharing this info, and please do let me know if I've understood correctly.

ssieb, 2021-07-06:
turn_off just sets the level to 0, so it's exactly the same.

prclm, 2021-07-07:
@Timmiej93 yes you got it right. Thank you for the explanation.

I also edited my formula in my previous post for easier understanding. Maybe you will take the changes over to your explanation? I think it would be less confusing.

> * [...] This value denotes the **un**-usable range of your light. If (with the normal setup) your light turns on at 20%, this value would be `0.2`.

I want to mention that this is logically true, but was not true in my case.

My light started flickering with 25% and below 24% was off. So by logic my conversion value would be `0.26`.
Actually I lost a lot of my brightness range with that value and i tested it with lower values. I came out with `0.023` for a perfect solution in my case. I have no idea why this is.

Maybe you would also try lower values. I like to know if this behavior is just with me.

Timmiej93, 2021-07-07:
Thank you both for getting back to me, I've incorporated both your replies in my comment above. Thanks again!

Timmiej93, 2021-07-07:
Accidentally closed this one, forgot that it was a feature request, and the feature still isn't there, just a workaround.

ssieb, 2021-07-07:
> My light started flickering with 25% and below 24% was off. So by logic my conversion value would be `0.26`.
> Actually I lost a lot of my brightness range with that value and i tested it with lower values. I came out with `0.023` for a perfect solution in my case. I have no idea why this is.

This is due to the gamma correction.  The output value is after that calculation is done. 0.25 ^ 2.8 ~= 0.021

closed 2022-07-05 by nagyrobi