esphome/feature-requests #1177: is_not_running

lancelon, 2021-03-27:
**Describe the problem you have/What new integration you would like**

I'm trying to perform a choice of two different actions depending on whether a button has been recently pressed. If it HAS been recently pressed, I want to 'escalate' the response. If it hasn't been recently pressed I'm happy to go with a low priority response. 

**Please describe your use case for this integration and alternatives you've tried:**

This is for an internal calling system in a warehouse. The end result is I want to cut down on staff stress - we don't always need a 'top level alert' response to a call for help. Hope this makes sense. However if the button is pressed within a short space of time of the original press it probably indicates a high priority! I've tried using `scripts `as pseudo timers and `is_script_running` but I can't figure out how to identify is_script_not_running
**Additional context**

Happy to go with an alternative approach, doesn't have to involve scripts. Someone suggested lambdas but I lack the nouse for this approach!

samnewman86, 2021-03-27:
This is what choose: is for.

lancelon, 2021-03-27:
Do I? How? With an else condition ?

samnewman86, 2021-03-27:
Yep.
If this.
Do this.
Else (no need to say if not this)
Do this

lancelon, 2021-03-27:
Thanks. By the way, what is "choice:" ?

closed 2022-07-01 by nagyrobi