esphome/feature-requests #1178: Support for PMSA003I Air Quality Sensor

shokwaav, 2021-03-28:
**Describe the problem you have/What new integration you would like**

I'm looking for support for the PMSA003I sensor, which can communicate over I2C.
Here's a link to the sensor - https://www.adafruit.com/product/4632 

**Please describe your use case for this integration and alternatives you've tried:**

It's an air quality sensor.

The existing integration only supports PMS5003, PMS7003 particulate matter sensors

**Additional context**

jorisa, 2021-07-11:
It looks like someone made an implementation: https://github.com/sjtrny/pmsa003i

Yet have to try myself

sjtrny, 2021-08-28:
It's now available in https://esphome.io/changelog/2021.8.0.html. This issue can be closed.

shokwaav, 2021-08-28:
> It's now available in https://esphome.io/changelog/2021.8.0.html. This issue can be closed.

Thanks for reminding me !

closed 2021-08-28 by shokwaav