esphome/feature-requests #1179: More Ezo Support

Morbidbystandard, 2021-03-28:
**Describe the problem you have/What new integration you would like**

Its very nice to have pH and rtd on ESPHome however there are some shortcomings. It would be nice to have calibration command entities that will issue i2c commands OTA to the ESP. Examples like "cal,mid,7", "cal,low,4", and "cal,high,10" for pH. I was able to change a few variables and get the EC probe running but it has the same issues. Lastly the flow sensor needs support and I'm sure people would like ORP as well. 

Pretty much what this post stated [https://github.com/esphome/feature-requests/issues/766#issue-634874366](url)

**Please describe your use case for this integration and alternatives you've tried:**

I've tried sending text inputs to the entity via node red with API failures.

**Additional context**

Tasmota has actually incorporated all sensors with calibration and I'd prefer to stay with ESPHome

serimi89, 2021-04-09:
I support your request, but (with my limited understanding) practically it seems a lot of commands to be implemented here in ESPHome.
I have have succeeded in sending ASCII commands to the EZO devices via Node-Red on Raspberry Pi connected directly via I2C (I can share the nodes if anyone is interested). Looking at ESPHome a straight-forward solution could be to create a MQTT topic that one could feed calibration commands in to and that would write to the I2C device/address specified? 
So in my case:
Node Red Text (in ASCII) => MQTT topic => ESPHome => I2C Write 
The output message then could be posted to the ESPHome log.

Any thoughts?

Morbidbystandard, 2021-04-09:
Thanks for the reply. You are correct. There are many commands unique to each sensor board. I'm currently tinkering with these to try to find a solution but my knowledge is very limited. You have an excellent clue to the path that needs to be taken. I will see what I can do.

One issue I have is, I'm running linux virtual machine on a windows 10 host. Unfortunately this setup seems to complicate everything.

serimi89, 2021-04-10:
Something is in the works.

kgalilio, 2021-04-20:
This will certainly be a great feature to have.
I have my PH and ORP sensors in the swimming pool pump room. The idea of removing the sensors and probes to calibrate them in my office where my PC is located, and then back is not practical. Being able to take the calibration solution to the pump room and click a button on my mobile device to calibrate is far more practical and convenient.

The following Arduino sketches are available from AtlasScientific:
**I2C**
ORP: [https://www.atlas-scientific.com/_files/code/ino_files/Ardunio-I2C-ORP-sample-code.zip](https://www.atlas-scientific.com/_files/code/ino_files/Ardunio-I2C-ORP-sample-code.zip)

PH:  [https://www.atlas-scientific.com/_files/code/ino_files/Ardunio-I2C-pH-sample-code.zip](https://www.atlas-scientific.com/_files/code/ino_files/Ardunio-I2C-pH-sample-code.zip)

RTD: [https://www.atlas-scientific.com/_files/code/ino_files/Ardunio-I2C-RTD-sample-code.zip](https://www.atlas-scientific.com/_files/code/ino_files/Ardunio-I2C-RTD-sample-code.zip)

I know very little about coding, so I am hoping that someone might find them useful and integrate the functionality into the ESPHome EZO library.

serimi89, 2021-04-20:
I have been working with one of the developers on this and currently there is a review process ongoing.

Morbidbystandard, 2021-04-21:
Very nice. Thank you. 

I wonder if there is a way to type a text input command like in the ardiuno serial monitor that can post to the various EZO sensors on the i2c bus.  It would definitely save time because there are sooooo many commands for these little guys. 

But when it comes down to it the calibration is the most important thing. It's not that crazy to command the EC stamp for TDS instead of uS or the flow stamp for current instead of total. Once they are set up you're good but calibration is ever changing.

maniscalcojohn, 2021-08-01:
Can add a request to support operating the dosing pump please??

auredor, 2021-08-24:
> I have been working with one of the developers on this and currently there is a review process ongoing.

Any update on the calibration review ? Thanks.

auredor, 2021-08-24:
> I have a pull request for this: [esphome/esphome#1865](https://github.com/esphome/esphome/pull/1865)

Thank you. Just seen the PR is blocked because of missing docs, right ? I guess we have to wait sometimes for that to be fixed.

ddragas, 2022-02-22:
any update on this PR ?

serimi89, 2022-02-22:
> any update on this PR ?

It’s online, right? 
https://esphome.io/components/sensor/ezo.html

gvdhoven, 2022-04-17:
@thomasdelaet can i help? With the pool season approaching and the ezo sensor boards at home, i was about to DIY something myself but having esphome support for pH and ORP (maybe EC in the future?) it would be perfect for both my pool and aquarium!

Morbidbystandard, 2022-04-17:
Hello,

Apparently, we need all the help we can get. It's been a year with no
progress. When it comes down to it, I'm a laymen.

I figured a way around this problem by running them in serial as a custom
sensor, and sending them commands with the UART switch.

It would be nice if the support page for EZO on esphome actually worked.
The temperature compensation is still a failure.

Thanks for your help

Assunto, 2022-04-28:
Hi All,

I was hoping there was a way to add the ability to have the ezo sensors work as a switch.  This is specifically for the peristaltic pumps that run off the i2c bus. They aren't really sensors, but they are ezo. 

I also can't find how to assign an i2c address to a switch. Is there a way to do that?

Thanks!

gvdhoven, 2022-06-15:
@Assunto there is a PR for EZO pumps pending review --> https://github.com/esphome/esphome/pull/3528

For the rest of the followers; i have revived an (old) PR to add not only calibration commands but also that you are able to issue your own commands to the sensor --> https://github.com/esphome/esphome/pull/3561

It is pending review by @ssieb after i have done some more testing.

Morbidbystandard, 2022-06-16:
> You have no idea how happy this makes me! Thank you *so* much!
>
> It looks like there are 4 active and progressing pull requests involving
> EZO sensors and pumps, #3528 #3561 #2133 and #2115 and I was wondering if
> there is anything a non-coder (me) can do to help push all these to the
> finish line, or how long you anticipate until they are ready to use?
>
> Again, @gvdhoven <https://github.com/gvdhoven> thank you so so so much

*grabs popcorn*

Thanks guys, this is great news.

gvdhoven, 2022-06-16:
It all is in the capable hands of SSieb, who i've contacted via Discord.

nagyrobi, 2022-07-01:
Work in progress, please continue discussion in the respective PRs comments section.

Morbidbystandard, 2022-07-01:
@serimi89 I hoping you are still subscribed. I would love to check out the nodes and sequence you have been using.

serimi89, 2022-07-01:
Hey @Morbidbystandard, yeah still subscribed.
I have moved my setup to a raspberry pi and directly read / write my PH / ORP probe through node red. 
Last year I was working closely with a developer from the ESPHome community (Senex) on a custom EZO sensor component that would also allow for commands to be send to the devices and not only read. I think he submitted it back then to be released but somehow it was never approved and the much simpler (just read) component is now live.

I still have the files on my computer and it was working like a charm with a previous version of ESPHome, but since I did an update it won't compile again. Would anyone want to have a look at it?

Morbidbystandard, 2022-07-08:
Good to hear from you @serimi89. I ran into the same problem. This is well beyond my skill level. I did open a ticket a while back and they were able to figure it out. I haven't had the time to adapt it to my personal issue but it looks very promising. Check it out.

https://github.com/esphome/issues/issues/2710#issuecomment-1006369558

serimi89, 2022-08-02:
Hey @Morbidbystandard 
I am trying to move back to ESPHome / ESP32 for managing my PH / ORP probes and naturally looked at this again.
I got it working with changing some lines of code in the eco.cpp file and I can successfully send custom commands to the device via MQTT, however I don't get any feedback back yet. 
For example: Sending command "L,0" (EZO LED OFF) will turn the LED off, however I don't get the feedback that is being send by the EZO Device.
I am looking into this, could be just something with my MQTT - I don't have access to device for a couple of days, I will keep you posted.

serimi89, 2022-08-03:
Sorry for the double post guys.

As I said above, with some small changes to https://github.com/esphome/esphome/pull/1684 the component ist working flawlessly again. I believe enhancing the existing EZO component with the feature of being able to send custom commands to the EZO module and reading its feedback (exactly what @Morbidbystandard requested when opening this thread and exactly my motivation to modify this component with @SenexCrenshaw), a significant increase in the usability of the component is achieved.

A lot of work has been put into this to make the changes and I am wondering how to push this?

Morbidbystandard, 2022-08-03:
@serimi89 thanks for the update. That's awesome you got it to work. I'm having more issues with ESPhome at this point and I'm considering throwing the whole thing away. I don't have time at the moment for anything other than work so the status quo remains

ssieb, 2022-11-09:
Resolved by https://github.com/esphome/esphome/pull/3561

closed 2022-11-09 by ssieb