esphome/feature-requests #1181: Telink TLSR8368 remote 2.4ghz add request

dckiller51, 2021-03-30:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

The use is to couple this computer screen lamp for which I have implemented an Esphome firmware.
With this original remote control you can:
 - Screen Light ON/OFF: Press Once
 - Increase Brightness: Turn Clockwise
 - Inscrease Color Temperature: Press & Turn Clockwise
 - Switch Mode: Double-click
 - Mood Light ON/OFF: Long Press
 - Descrease Brightness: Turn Anticlockwise
 - Descrease Color Temperature: Press & Turn Anticlockwise

**Additional context**

I put the SDK and notices on the link below

https://github.com/dckiller51/esphome-yeelight-led-screen-light-bar

[image]

K-4U, 2022-07-10:
I'd love to have this as well. I haven't been doing C++ for ages so unfortunately i cannot help develop :(

K-4U, 2022-07-16:
So i was able to do it:
https://github.com/K-4U/custom_components/tree/main/components/yeelight_remote