esphome/feature-requests #1187: ST7796 4.0" Display Support

dwaterford75, 2021-04-07:
**Describe the problem you have/What new integration you would like**

So I have a 4.0" TFT Touch Display based on the ST7796 Driver which supports touch and runs on the SPI bus.
Is there any way to get support for this device in ESPHome?

**Please describe your use case for this integration and alternatives you've tried:**

I have tried the ILI9341 display configuration for this device and I get information on the screen but the size is wrong and it doesn't fill the whole screen.  If I put a 2.4" ILI9341 device on, its perfect but I want a bigger display.

**Additional context**

richardklingler, 2021-05-17:
What exact display is it? Which touchscreen IC?

I have a ST7796S and not ST7796  driver in the works...but it is not a 4" display either.

dwaterford75, 2021-05-17:
[image]
[image]
[image]
[image]

From what I am seeing on other sites, this is a ST7796S and not ST7796.

Are you programming your unit via ESPHome?  If so, could you maybe outline connections and sketch?  Thanks for the help!

dwaterford75, 2021-05-17:
[image]

This display definitely works with the ESP32, just not sure how to get it working within ESPHome ... Looks to me like it all boils down to a driver issue.  If I try run it as a ili9341 with TFT_2.4 it works but the screen limits X&Y are as per the 2.4 and not the 4.0.

dwaterford75, 2021-05-17:
Apologies - I closed the issue by mistake.

dwaterford75, 2021-05-17:
[image]
[image]

display:
  - platform: ili9341
    model: m5stack
    led_pin: GPIO22
    cs_pin: GPIO5
    dc_pin: GPIO16
    reset_pin: GPIO17
    rotation: 90
    id: my_display

Colours wrong, orientation wrong and not able to use the whole screen BUT doing stuff.

dwaterford75, 2021-05-17:
display:
  - platform: ili9341
    model: TFT 2.4

Changes the display to:

[image]
[image]

kd7gab, 2021-11-10:
I would really like to be able to utilize the WT32-SC01 units with ESPHome. Has this request simply been abandoned? I see the repo linked earlier on no longer exists...

poundy, 2022-01-03:
can't really help with WT32-SC01 (don't have one) but I do note that the other person commenting here does take sponsorships to publish devices... see their github https://github.com/richardklingler/esphome/

poundy, 2022-01-03:
Now you got me looking though. Where did you get the 4" version from? I can't seem to find one.

I've played a bit with ILI9341 displays in the past, and just a touch in ESPHome, but a 4" display could be a nice option - my go-to has always been the 240x320 2.8" displays as they seemed to be the right balance in screen realestate and resolution.

poundy, 2022-01-03:
all the code in esphome hard-codes the display sizes to 320x240 resolutions. This means at a minimum you would need to revise that to be able to use the extra screen pixels. I've asked on Discord, and I've got some hardware coming, perhaps once that arrives I'd be able to help test. 
https://github.com/esphome/esphome/blob/dev/esphome/components/ili9341/ili9341_display.cpp is the display driver.

poundy, 2022-01-05:
What would be great is for anyone following along here and who has one of these 320x480 displays to thoroughly test out the ILI9341 2.4" option within esphome and report if, besides not being able to use the whole screen, everything else works as expected?  This will test if it's just the screen resolution that needs adjusting (which uses more memory for the screen buffer - not a big deal on an ESP32 tho, but this will never work on an ESP8266, even the 240x320 doesn't). Hopefully it'll be suitably compatible with the ILI9341 code

poundy, 2022-02-02:
Hardware arrived. Can't repro the semi-working experience @dwaterford75 had - any chance you can dig out pins and setup that you tested this with? I've started looking at code

poundy, 2022-04-04:
@Coolie1101 I think your problem has given me a slightly different angle on how to potentially address this request... 

There seems to be two significant barriers. First, the ST7789V code that is in esphome today seems to be fixed pixel dimensions (small !) so even if the driver worked for your display, it won't handle your resolution. Second, the ILI9341 code that kinda works on ST7789's (M5Stack option like you used above) is also at a fixed pixel dimension as well, 240x320, but doesn't handle all hardware specifics like mirroring the display like you need.  

To address this for 480x320 resolution screens (there is no supported option here at all at the moment) may need a massive change in structure so there's more screen size options for everyone.  

For you however it could be worth trying the edit in https://community.home-assistant.io/t/esphome-st7789v-modifications-to-use-a-240x320-pixel-screen/256347/10 that should be able to confirm if the ST7789V driver handles the setup of the screen better than the ILI9341 does by giving you correct colours and not mirrored display.  

The common 480x320 displays I found are also using an ST7789V like yours, so to properly support them needs thinking through how best to do that... I might kick that back to the actual esphome team to discuss.

Coolie1101, 2022-04-04:
> could be worth trying the edit

Hey @poundy, thanks for following up, your recommendation seems simple enough, but what exactly do I change in those files?, just the resolution?

> confirm if the ST7789V driver handles the setup of the screen better than the ILI9341 does

Though not ideal, the colors I can work around, but if we had a flip option, that would take care of the mirror effect.

poundy, 2022-04-04:
I didn't look at the thread in much depth, but for a start I'd be looking at width and height params and seeing how they're used. More than that, and it becomes a somewhat-developer-delve-and-solve problem and I don't know how comfortable you are going deep... but hey, can't hurt either ;-)  I think you're right in that it'll be necessary to change the resolution where you note, but without looking at the code I don't know what the X/Y offset values are for or how they have to change.

I saw this update just as I was pinging the #dev channel on discord. Will see what they think and I'll post the link to that discussion once complete

Coolie1101, 2022-04-04:
So, I tried adding the edited files to "custom_components/st7789v/", but Esphome shows an error "Platform not found: 'display.st7789v' when configured, once I rename the "st7789v" directory to something else, the error goes away.

nagyrobi, 2022-07-24:
Try now https://github.com/esphome/esphome/pull/3651

clowrey, 2022-08-18:
> Try now [esphome/esphome#3651](https://github.com/esphome/esphome/pull/3651)

I just tried to get my WT32-SC01 display which is 480x320 using ST7796S to work using the updated ST7789V driver which supports higher resolutions. 

At first it would compile but then give "unable to allocate buffer errors" until I changed the "board" to one with 16mb and the PSRAM that this board has:

```
esp32:
  board: m5stack-core2
  framework:
    type: arduino
```
It boots without errors now, but I am getting nothing but a blank white display on the LCD. 

Then I tried it using the ili9341 display driver, and it actually displays text etc - and actually upright in the correct orientation, but not covering the whole display etc. 

```
spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13
  
display:
  - platform: ili9341
    model: TFT 2.4
    cs_pin: GPIO15
    dc_pin: GPIO21
    reset_pin: GPIO22
    led_pin: GPIO23
```

So this is promising at least ;) it appears that this ili9341 driver will just need to be updated similar to how the ST7789V driver was and then the WT32-SC01 will be supported, then would just need the FT6336U capacitive touch chip too ideally, but I was more interested in just displaying data vs user input...

dwaterford75, 2022-09-04:
Hi All,
Did anyone make any progress on this issue?
Thanks in advance.

poundy, 2022-09-08:
@dwaterford75 no I haven't tested anything. From your comment above, it seems the WT32-SC01's controller chip most closely matches the ili9341 and I'd agree that's the most promising area to pursue, but in my initial reviews it was beyond my programming ability to appropriately refactor it to suit. The [delta report for this pull request ](https://github.com/esphome/esphome/pull/3651/files/93f2fd4ea986dba5cc6b568aa1b0d424cbd99a8f) however might shed more light on the form of the change, that might make it simpler for me to follow the bouncing ball and get it to work...

clowrey, 2022-09-21:
It works!! 
Using my very hacked up "SD7796" (all names still ili9341) component here: https://github.com/clowrey/esphome/tree/dev/esphome/components/ili9341

#configure for PSRAM support which is needed for 320x480 buffer size
esp32:
  board: m5stack-core2
  framework:
    type: arduino

 # override existing ili9341 component - not necessary once someone(maybe me if I ever have time) makes a proper SD7796 component using this example
external_components:
  - source: github://clowrey/esphome@dev
    components: [ ili9341 ] 
    refresh: 1min

Display config section for the WT32-SC01 board is: 

spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13

display:
  - platform: ili9341
    model: TFT 2.4
    cs_pin: GPIO15
    dc_pin: GPIO21
    reset_pin: GPIO22
    led_pin: GPIO23
    lambda: |-
      it.fill(COLOR_OFF); // Turn the whole display off.
      it.rectangle(0,  0, it.get_width(), it.get_height(), id(my_blue)); // entire screen border
      for (int i = 1; i <= 105; ++i) {
        it.printf((i * 5), (i * 10), id(helvetica_12), id(my_green), TextAlign::CENTER, "Hello World %i", i); // fill screen with text
      }

[image]

gpambrozio, 2022-09-26:
Thanks @clowrey for figuring this out. I created [a PR](https://github.com/esphome/esphome/pull/3848) to add this to esphome. The code is pretty much the same but packaging to be used with esphome in a way that works with the rest of the project without any hacking. Example yaml in PR.

clowrey, 2022-09-28:
@gpambrozio Great! yup that looks like what I would have done if I had more time! glad you got it refactored properly. Cheers

clowrey, 2022-09-28:
Now we just need the FT6336U touch IC driver ;)

gpambrozio, 2022-10-03:
@clowrey still a draft but if you want to test it and report back I'd appreciate it: https://github.com/esphome/esphome/pull/3865

clowrey, 2022-10-04:
@gpambrozio awesome! I'll be away from computer for a few more days but will try it out when I can! Looking forward to that, this will complete being able to use these capacitive LCDs for info + control displays in ESPhome

gpambrozio, 2022-10-05:
My display code was now added to https://github.com/esphome/esphome/pull/3795 so I closed my pr.

shahargross85, 2023-02-25:
Thanks for this!
I have managed to change the driver to increase the resolution but up to 320x360. Above that resolution, the ESP crashes.
I think it is because of memory shortage.
I'm using the WROOM 32. 

I see you mentioned support of PSRAM, is this my issue?
If so, how do I know if I have a PSRAM on my HW, how do I use it?

nielsnl68, 2023-02-26:
> I see you mentioned support of PSRAM, is this my issue?
> If so, how do I know if I have a PSRAM on my HW, how do I use it?

@shahargross85 most likely. ili9xxx will report of you have PSRAM or not.

shahargross85, 2023-02-27:
I'm using this configuration:

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

Which ESP do you recommend to work with which support 320x480 screen resolution?

See the logs below. I don't see any information about PSRAM:

[09:10:09][I][app:102]: ESPHome version 2023.2.0-dev compiled on Feb 25 2023, 23:50:07
[09:10:09][C][wifi:504]: WiFi:
[09:10:09][C][wifi:362]:   Local MAC: 80:64:6F:C4:95:0C
[09:10:09][C][wifi:363]:   SSID: [redacted]
[09:10:09][C][wifi:364]:   IP Address: 10.100.102.215
[09:10:09][C][wifi:366]:   BSSID: [redacted]
[09:10:09][C][wifi:367]:   Hostname: 'screen'
[09:10:09][C][wifi:369]:   Signal strength: -58 dB ▂▄▆█
[09:10:09][C][wifi:373]:   Channel: 5
[09:10:09][C][wifi:374]:   Subnet: 255.255.255.0
[09:10:09][C][wifi:375]:   Gateway: 10.100.102.1
[09:10:09][C][wifi:376]:   DNS1: 10.100.102.1
[09:10:09][C][wifi:377]:   DNS2: 0.0.0.0
[09:10:09][C][logger:293]: Logger:
[09:10:09][C][logger:294]:   Level: DEBUG
[09:10:09][C][logger:295]:   Log Baud Rate: 115200
[09:10:09][C][logger:296]:   Hardware UART: UART0
[09:10:09][C][spi:101]: SPI bus:
[09:10:09][C][spi:102]:   CLK Pin: GPIO18
[09:10:09][C][spi:103]:   MISO Pin: GPIO19
[09:10:09][C][spi:104]:   MOSI Pin: GPIO23
[09:10:09][C][spi:106]:   Using HW SPI: YES
[09:10:09][C][ledc.output:114]: LEDC Output:
[09:10:09][C][ledc.output:115]:   Pin GPIO15
[09:10:09][C][ledc.output:116]:   LEDC Channel: 0
[09:10:09][C][ledc.output:117]:   Frequency: 1000.0 Hz
[09:10:09][C][ili9341:029]: ili9341
[09:10:09][C][ili9341:029]:   Rotations: 0 °
[09:10:09][C][ili9341:029]:   Dimensions: 320px x 340px
[09:10:09][C][ili9341:030]:   Reset Pin: GPIO22
[09:10:09][C][ili9341:031]:   DC Pin: GPIO4
[09:10:09][C][ili9341:033]:   Update Interval: 1.0s
[09:10:09][C][homeassistant.time:010]: Home Assistant Time:
[09:10:09][C][homeassistant.time:011]:   Timezone: 'IST-2IDT,M3.4.4/26,M10.5.0'
[09:10:09][C][xpt2046:141]: XPT2046:
[09:10:09][C][xpt2046:143]:   IRQ Pin: GPIO27
[09:10:09][C][xpt2046:144]:   X min: 193
[09:10:09][C][xpt2046:145]:   X max: 3864
[09:10:09][C][xpt2046:146]:   Y min: 238
[09:10:09][C][xpt2046:147]:   Y max: 3924
[09:10:09][C][xpt2046:149]:   Swap X/Y: NO
[09:10:09][C][xpt2046:150]:   Invert X: NO
[09:10:09][C][xpt2046:151]:   Invert Y: YES
[09:10:09][C][xpt2046:153]:   threshold: 400
[09:10:09][C][xpt2046:154]:   Report interval: 4294967295
[09:10:09][C][xpt2046:156]:   Update Interval: 0.050s
[09:10:09][C][captive_portal:088]: Captive Portal:
[09:10:09][C][mdns:108]: mDNS:
[09:10:09][C][mdns:109]:   Hostname: screen
[09:10:09][C][ota:093]: Over-The-Air Updates:
[09:10:09][C][ota:094]:   Address: screen.local:3232
[09:10:09][C][ota:097]:   Using Password.
[09:10:09][C][api:138]: API Server:
[09:10:09][C][api:139]:   Address: screen.local:6053
[09:10:09][C][api:141]:   Using noise encryption: YES

nielsnl68, 2023-02-27:
You could try to add `psram:`  to your yaml file to see of there is psram. 
BTW within the ili9xxx platform component (driver) this will be automatically enabled.

shahargross85, 2023-02-27:
Thanks. Sadly no PSRAM (or am I not enabling it?):
[11:24:21][C][psram:020]: PSRAM:
[11:24:21][C][psram:021]:   Available: NO

which board ESP should I use for such an application?
Can I use higher resolution without a PSRAM?

nielsnl68, 2023-02-27:
You can check this list and find a module here: https://www.espressif.com/en/products/modules

jcgonzalezdelgado, 2023-05-08:
> 

Did you make it work without PSRAM? If not, which board did you get? I ordered a few 3.5 480x320 screens and wanted to make sure I don't need new esp32 boards

shahargross85, 2023-05-08:
Hi,

No, I had to use a ESP with an external PSRAM, otherwise the device crashed as it didn't have enough memory
When I switched from WROOM to WROOVER it worked

netmindz, 2023-05-20:
@clowrey  If I use ili9341 then it moans that it's been renamed but just get a flashing black and white full screen on boot, then nothing on my WT32-SC01

If I follow the suggestion to use ili9xxx then I do get output, but only on approximately 70% of the display

netmindz, 2023-05-20:
@clowrey 

[C][spi:023]: Setting up SPI bus...
[E][display:109]: Could not allocate buffer for display!
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400894bc  PS      : 0x00060230  A0      : 0x800d8a0c  A1      : 0x3ffb2660  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00025800  A5      : 0x00000000  
A6      : 0x000001df  A7      : 0x00002580  A8      : 0x8016bba1  A9      : 0x3ffb2630  
A10     : 0x000001e0  A11     : 0x3ffb29a4  A12     : 0x00000040  A13     : 0x3f401635  
A14     : 0x00000008  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400894bc  LEND    : 0x400894c7  LCOUNT  : 0x0000257f  

Backtrace:0x400894b9:0x3ffb26600x400d8a09:0x3ffb2670 0x400d8a43:0x3ffb2690 0x400decf6:0x3ffb26b0 0x4016bf61:0x3ffb2700 0x400de934:0x3ffb2720 0x400e00a6:0x3ffb2750 0x400ebc3e:0x3ffb2820

netmindz, 2023-05-20:
adding psram: resolved the issue