esphome/feature-requests #1191: Time Offset

RebbePod, 2021-04-15:
**Describe the problem you have/What new integration you would like**
When making an automaton using sunrise or sunset. it would be really great to have an offset option, so for example if i want the light in front of my house to go on 30 min before sunset.

The feature should allow for offsets above or below a sunrise / sunset time.

**Please describe your use case for this integration and alternatives you've tried:**
My use case is as mentioned above.

**Additional context**
Home Assistant has this feature, so I assume it shouldn't be to difficult to port over.

fraxinas, 2022-01-10:
I second this feature request.

I've tried working with different elevations, but this is highly depended on the season

RebbePod, 2024-03-20:
I've achieved it using a function running every minute. But that's not ideal

twilliams0805, 2024-06-04:
This would be a very useful feature for a lot of automations. Would be great to have it available directly via ESPHome.