esphome/feature-requests #1193: Hello, please add support for electrolux air conditioners to the infrared climate control. Thanks!

Valerich1971, 2021-04-16:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

SebastianBar, 2021-11-03:
@Valerich1971 Are you sure you need a specific implementation for Electrolux?
I have an Electrolux branded AC, and it fully works by using `platform: tcl112`

Valerich1971, 2021-11-04:
thanks, I've tried all platforms, but I'll try again with tcl112 04.11.2021, 02:57, "Sebastián Barrenechea" ***@***.******@***.*** Are you sure you need a specific implementation for Electrolux?I have an Electrolux branded AC, and it fully works by using platform: tcl112

csomyy, 2023-02-01:
Upping this! I got an Electrolux climate too and would control it via IR. Tried every single protocols but cannot get it to work.

ssieb, 2023-02-02:
Someone will have to get protocol information for the device.