esphome/feature-requests #1196: Please support m5stack-coreink board

ax42, 2021-04-20:
**Describe the problem you have/What new integration you would like**

Support for https://docs.m5stack.com/#/en/core/coreink

**Please describe your use case for this integration and alternatives you've tried:**

As per the platformio documentation, the board is supported: https://docs.platformio.org/en/latest/boards/espressif32/m5stack-coreink.html#board-espressif32-m5stack-coreink

When I put `m5stack-coreink` as the board type in the esphome config, I get an 'could not find board' error.

**Additional context**

Currently using esphome 1.16.2 (Home Assistant addon)

ax42, 2021-04-20:
[image]

koenvervloesem, 2021-04-20:
What if you try the following:

```
esphome:
  name: "$[devicename}"
  platform: esp32
  board: m5stack-core-esp32
  platformio_options:
    platform: espressif32
    board: m5stack-coreink
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L
```

ax42, 2021-04-20:
Thanks. It allows me to proceed to the compilation step, but that has an 'unknown board' error.

[image]

(for anyone copy-pasting the above, there is a small typo, should be `name: "${devicename}"` (note the curly braces)

koenvervloesem, 2021-04-20:
Is your PlatformIO the newest version?

ax42, 2021-04-20:
It's the one within the home assistant plug-in. How do I check / update?

koenvervloesem, 2021-04-20:
Sorry, I'm not sure. I'm using ESPHome and PlatformIO standalone, not as a Home Assistant add-on.

ax42, 2021-04-20:
I've just tried with the esphome/esphome:dev docker image (v1.18.0-dev), and get the same error on compilation.

koenvervloesem, 2021-04-20:
Weird. I don't have a CoreInk device, but I've used the same approach successfully for another board that is supported in PlatformIO but not in ESPHome.

asteele4, 2021-07-24:
I'd also like to see this device supported.  As well as the board, support is needed to the e-paper display (gdew0154M09).

ax42, 2022-07-15:
Board now seems to be supported (the code compiles) using the sketch below, any ideas on how to get the e-paper display working (as per previous comment)?

```
esp32:
  board: m5stack-coreink
  ```

ax42, 2022-11-10:
Anything new here?

witasekl, 2023-02-13:
Hi @ax42,
I've implemented external component for the e-paper display. Check my repo here: <https://github.com/witasekl/gdew0154m09>