esphome/feature-requests #1202: Read Value From another esp

DotumZane, 2021-05-01:
**Describe the problem you have/What new integration you would like**

I have one esp reading and reporting air pressure, and I have another that I would like to display the system pressure on a 7 segment display. 
**Please describe your use case for this integration and alternatives you've tried:**

there's no easy way for two esp devices to talk to each other.
**Additional context**

randybb, 2021-05-02:
You can use https://esphome.io/components/sensor/homeassistant.html

grea09, 2022-01-21:
I was searching for that because of a use case I have. My home assistant instance tends to lose connectivity. My heater installation is quite sensitive and I got pumps and valves on different esp. I have a very important need for the pump to never ever be active while the valves are closed and that means to directly fetch sensor value from the other esp.

I understand that is a rare use case but I commented so that it is known for others.

randybb, 2022-01-21:
Currently [web api ](https://esphome.io/web-api/index.html)+ [http request](https://esphome.io/components/http_request.html) that relays on wifi. Similarly it should be possible to use remote_transmitter and remote_receiver.
Modbus, CAN and BLE are running in master mode. esp_now support will come soon.

closed 2022-06-29 by nagyrobi