esphome/feature-requests #1214: Save all the compiled firmware to a common directory with the name of the node.

Korte68, 2021-05-07:
**Describe the problem you have/What new integration you would like**
Save all the compiled firmware to a common directory with the name of the node (kitchen1.bin,hall3.bin....)

**Please describe your use case for this integration and alternatives you've tried:**
in case of lot of node, it would be easier to find the compiled binary, and if we need to upload over USB, it is easy to find and identify the node.

**Additional context**

nagyrobi, 2021-10-31:
Another approach: create a link in the ESPHome dashboard: _Download all the current binaries_. Clicking on this would grab all the cached binary files + yaml configs for all the nodes in the system, name them correctly (as above) and zip them in a common archived file prompted for download.

A simple and easy way to make backups.

nagyrobi, 2022-07-07:
Duplicate of https://github.com/esphome/feature-requests/issues/1423

closed 2022-07-07 by nagyrobi