esphome/feature-requests #1216: Add support for Adafruit_DS3502

paracetamol32, 2021-05-09:
**Describe the problem you have/What new integration you would like**
Add support for the Adafruit DS3502 I2C Potentiometer library ,Thank you for your excelent job with ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**
Use the DS3502 I2C Potentiometer to control Viridian EV EPC (cf: http://evbitz.uk/EVSE_Kit_-_Type_2_files/EPC%20Manual%20V1.2_050816.pdf ).
Viridian use a variable resistor to control output power for charging electic car

**Additional context**

spec here : https://github.com/adafruit/Adafruit_DS3502

boozeman, 2022-03-21:
Any progress for this? 

DS3502 is handy sensor for voltage controlled EC-fans (0-10v with 10K pot)

pascaltippelt, 2022-06-27:
I would also really appreciate an DS3502 integration.

lordratner, 2023-01-23:
Adding my name to the list of people wanting support for this. Thanks for everything you do!

inspectorgrowadmin, 2023-05-10:
Also would like this integrated

Lou-Menz, 2023-06-03:
Yes, please add this.

I need to control a whole house filter fan by a variable resistance through the controller board.

PWM directly to fan motor won't work.

Thanks

lordratner, 2023-06-03:
> Yes, please add this.
> 
> I need to control a whole house filter fan by a variable resistance through the controller board.
> 
> PWM directly to fan motor won't work.
> 
> Thanks

> very useful component to control 0-10V. By now I use it with mysensors and arduino for my 0-10V Fans, look forward to move to esphome as soon as it will be supported 👍

Just a heads up, I have this working on ESPHome with this: https://github.com/esphome/feature-requests/issues/1216#issuecomment-1575127065

It's quite simple to adapt to your needs, until there is native support.

Lou-Menz, 2023-06-03:
Thank you lordratner!

UserRS232, 2023-06-08:
@lordratner  Thanks, i have trouble getting them work. Can you please write a short guide on how to do this? I'm fairly new to the ESPHome universe

lordratner, 2023-06-08:
Put his file, DS3502.h in /config/esphome

```
#include "esphome.h"
#include <Adafruit_DS3502.h>

#define WIPER_VALUE_PIN A0

class DS3502_Component : public Component, public FloatOutput
{
public:

    int I2C_Address;
    Adafruit_DS3502 ds3502;

    DS3502_Component(int address = DS3502_I2CADDR_DEFAULT) {
        I2C_Address = address;
        ds3502 = Adafruit_DS3502();
        ESP_LOGI("custom", "Initializing <%i>", I2C_Address);
    }

    void setup() override {
        if (!ds3502.begin(I2C_Address))
        {
            ESP_LOGI("custom", "<0x%X> Couldn't find DS3502 chip", I2C_Address);
            while (1);
        }
        ESP_LOGI("custom", "<0x%X> Found DS3502 chip", I2C_Address);
    }

    void write_state(float state) override {
        int mappedState = map(int(state*100), 0, 100, 0, 127);
        ESP_LOGI("custom", "Setting wiper of <%i> to <%i>", I2C_Address, mappedState);
        ds3502.setWiper(mappedState);
        int wiperValue = ds3502.getWiper();
        ESP_LOGI("custom", "--> Wiper value of <%i> is now <%i>", I2C_Address, wiperValue);
    }
};
```
In your device yaml, you need i2c

```
i2c:
  sda: D4
  scl: D3
  scan: false
```
An output:

```output:
  - platform: custom
    type: float
    lambda: |-
      auto digipot = new DS3502_Component(0x28);
      App.register_component(digipot);
      return {digipot};
    outputs:
      - id: dc_fan_output

### Only needed for climate devices
  - platform: template
    id: fan_control
    type: float
    write_action:
      - fan.turn_on:
          id: datacenter_fan
          speed: !lambda return state * 100;
###
```
and a fan:

```
fan:
 - platform: speed
   id: datacenter_fan
   output: dc_fan_output
   name: Fan
```

Note that I have two outputs. The first is needed in a normal config, and will give you a speed fan in Home Assistant. The second is for if you plan to use the fan as a cooler in a climate system, such as a PID, since it can only control an output component, and not a fan device. So the second output device accepts the control input from the PID and send the value to the fan device with `write action`

Additionally, if you want you fan to have a permanent minimum speed so it is always running, but still with variable speed, change this line in the DS3502.h file:

`        int mappedState = map(int(state*100), 0, 100, 0, 127);`

Change the second '0' to whatever you want your minimum to be (on a scale of 0-127). So if you want the fan to run at 10% when Home Assistant sets it to off/0, then change the second '0' to 13. Consider this an advanced feature, so make sure you know what you're trying to accomplish. If that number is anything other than '0' the fan will never stop.

UserRS232, 2023-06-08:
wow, super quick response.  it works!!  thank you very much for your great answer

Lou-Menz, 2023-06-08:
I'd like to add my THANKS too!

WarmeBrille, 2023-10-11:
Hello everyone,

I am just starting with the whole esp home and breakout board topic and I'm wondering if I still need a D1 Mini or ESP8266 board to run a the digital poti.
Or if I can just wire it to the SDA an SCL pins of the pi an connect as it is given here:
https://learn.adafruit.com/ds3502-i2c-potentiometer/python-circuitpython

Thanks in advance!

robinostlund, 2024-03-09:
@lordratner i just updated my esphome to latest version and it stopped responding, guess it is something with the external library. Have you seen any issues with it lately? :)

lordratner, 2024-03-14:
> @lordratner i just updated my esphome to latest version and it stopped responding, guess it is something with the external library. Have you seen any issues with it lately? :)

No issue with mine, just updated to the most recent version.

robinostlund, 2024-04-13:
@lordratner sorry for late reply, but are you using esp 8266 or esp32? I am using esp32 so wondering if that could be the issue.

marci144, 2024-12-07:
Hello @lordratner 

I have tried your solution on an ESP32 C3 
Added all the includs required, and when I compile, I got: 

```
Compiling .pioenvs/test-potenziometro/src/main.cpp.o
In file included from src/main.cpp:34:
src/DS3502.h:7:1: error: expected class-name before '{' token
 {
 ^
src/DS3502.h:28:10: error: 'void DS3502_Component::write_state(float)' marked 'override', but does not override
     void write_state(float state) override {
          ^~~~~~~~~~~
*** [.pioenvs/test-potenziometro/src/main.cpp.o] Error 1
```
The file compiles but clearly fails if I remove the function setup() and the second inheritance of class DS3502_Component (FloatOutput).

Any advice you can give me?
Thanks in advance 
Marcello

p1ngb4ck, 2025-02-02:
Hey everybody ..
just in case this could be helpful to anybody here:
I hust finished work for v 1.0 of my MCP4461 quad i2c potentiometer/rheostat component. PR is open and I hope it will get merged.
The MCP4461 is available in multiple (5k/10k/50k/100k) total resistance versions and could be exactly what you need maybe.
I personally use the 100k variant to dim MeanWell powersupplies, having 3in1 dimming functionality.
MCP4461 can be controlled using 256 taps (/steps), it`s internally built using resistor ladders. So eg. for the 10k version, you`d have possible net values of [0-256] * (<max resistance>/256). In my case the actual max-resistance of my 100k version is 96.7k or sth., but my multimeter is cheap and the chip could be a clone. Close enough to my needs though anyway and I could add a static resistor if I wanted, as those powersupplies have a 6% on-treshold anyhow.

marci144, 2025-02-28:
Yes !!
Happy to give it a try !

Tks
Marcello

Ylianst, 2025-07-20:
I used the @lordratner solution above for a long time and it worked great, but with the latest ESPHome, `platform: custom` is no longer supported and so, I now need to find an alternative.