esphome/feature-requests #1217: Support ESP32-C3

stintel, 2021-05-10:
Support ESP32-C3

The ESP32-C3 is a RISC-V based ESP. It would be nice if esphome supported this.

ESP32-C3-DevKitM-1 can be ordered from e.g. [Mouser](https://www.mouser.com/ProductDetail/Espressif-Systems/ESP32-C3-DevKitM-1?qs=pUKx8fyJudB1sOWbbEnGFw%3D%3D). Estimated shipping date for mine is 17/05. I'm willing to donate 1 or 2 if needed.

markusg1234, 2021-05-13:
Got mine yesterday and have it working in Arduino IDE using https://github.com/espressif/arduino-esp32#master
Would be great if it can be added to ESPHome.

Thankyou.

misery, 2021-05-17:
@stintel I created a PR (https://github.com/esphome/esphome/pull/1433) to support my ESP32-S2 for 1.18.0. https://github.com/espressif/arduino-esp32/releases/tag/2.0.0-alpha1 supports both now. So maybe it should work, too.

Could you try this? Does it work?
You need to use current dev branch: https://github.com/esphome/esphome

```
esphome:
  name: esp32c3demo
  platform: ESP32
  board: nodemcu-32s
  platformio_options:
    board_build.mcu: esp32c3
    platform_packages: framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
 ```

```
esphome your_c3.yaml compile
```

I don't have a C3 yet. So please post the compile errors here. :-)

stintel, 2021-05-17:
> I don't have a C3 yet. So please post the compile errors here. :-)

Still waiting for mine, and the estimated shipment date changed today:
`* We will contact you as soon as we obtain an estimated shipment date from the manufacturer.`

But thanks in advance, and I'll definitely try it out once I get my C3s.

markusg1234, 2021-05-18:
Hi, I tried your suggestion and got the following with an error of failed.

INFO Reading configuration /config/esphome/esp32-c.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/esp32-c
Processing esp32-c (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.7856de7 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
Library Manager: AsyncTCP-esphome @ 1.2.2 has been installed!
Library Manager: Installing ESPAsyncWebServer-esphome @ 1.2.7
Library Manager: ESPAsyncWebServer-esphome @ 1.2.7 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing ESPAsyncTCP-esphome
Library Manager: ESPAsyncTCP-esphome @ 1.2.3 has been installed!
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <AsyncTCP-esphome> 1.2.2
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/esp32-c/.pioenvs/esp32-c/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32c3-elf-g++: not found
Compiling /data/esp32-c/.pioenvs/esp32-c/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32c3-elf-g++: not found
Compiling /data/esp32-c/.pioenvs/esp32-c/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: xtensa-esp32c3-elf-g++: not found
*** [/data/esp32-c/.pioenvs/esp32-c/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /data/esp32-c/.pioenvs/esp32-c/src/esphome/components/api/api_server.cpp.o
*** [/data/esp32-c/.pioenvs/esp32-c/src/esphome/components/api/api_pb2.cpp.o] Error 127
*** [/data/esp32-c/.pioenvs/esp32-c/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
sh: 1: xtensa-esp32c3-elf-g++: not found
*** [/data/esp32-c/.pioenvs/esp32-c/src/esphome/components/api/api_server.cpp.o] Error 127
========================= [FAILED] Took 18.02 seconds =========================

misery, 2021-05-18:
Hi, thanks.... I tried to compile it myself now.

Found in https://github.com/platformio/platform-espressif32/issues/497 that someone added the toolchain for it.
```
  platform: ESP32
  board: nodemcu-32s
  platformio_options:
    board_build.mcu: esp32c3
    platform: https://github.com/Jason2866/platform-espressif32.git#feature/arduino-c3
    platform_packages: framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
```

I have some compile errors now. I will look into it later.
Initial error fixed: https://github.com/esphome/esphome/pull/1798

misery, 2021-05-18:
Fixed some compile errors. Please check if this helps to run a simple test. I dont' have any C3 here to execute the binary.

Known issue:
- captive_portal
- web_server

```
.piolibdeps/esp_demo_c3/ESPAsyncWebServer-esphome/src/AsyncWebSocket.cpp: In member function 'IPAddress AsyncWebSocketClient::remoteIP()':
.piolibdeps/esp_demo_c3/ESPAsyncWebServer-esphome/src/AsyncWebSocket.cpp:840:28: error: call of overloaded 'IPAddress(unsigned int)' is ambiguous
         return IPAddress(0U);
                            ^
In file included from /home/andre/.platformio/packages/framework-arduinoespressif32@src-cba3def1496a47e6af73c0b73bd2e13c/cores/esp32/Arduino.h:153,
                 from .piolibdeps/esp_demo_c3/ESPAsyncWebServer-esphome/src/AsyncWebSocket.cpp:21:
/home/andre/.platformio/packages/framework-arduinoespressif32@src-cba3def1496a47e6af73c0b73bd2e13c/cores/esp32/IPAddress.h:51:5: note: candidate: 'IPAddress::IPAddress(const uint8_t*)'
     IPAddress(const uint8_t *address);
     ^~~~~~~~~
/home/andre/.platformio/packages/framework-arduinoespressif32@src-cba3def1496a47e6af73c0b73bd2e13c/cores/esp32/IPAddress.h:50:5: note: candidate: 'IPAddress::IPAddress(uint32_t)'
     IPAddress(uint32_t address);
     ^~~~~~~~~
/home/andre/.platformio/packages/framework-arduinoespressif32@src-cba3def1496a47e6af73c0b73bd2e13c/cores/esp32/IPAddress.h:29:7: note: candidate: 'constexpr IPAddress::IPAddress(const IPAddress&)'
 class IPAddress: public Printable
```

Another TODO with GPIO usage:
```
/home/andre/.platformio/packages/toolchain-riscv32/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/esp_demo_c3/libFrameworkArduino.a(wiring_pulse.c.o): in function `pulseIn':
/home/andre/.platformio/packages/framework-arduinoespressif32@src-cba3def1496a47e6af73c0b73bd2e13c/cores/esp32/wiring_pulse.c:38: undefined reference to `xthal_get_ccount'
/home/andre/.platformio/packages/toolchain-riscv32/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/andre/.platformio/packages/framework-arduinoespressif32@src-cba3def1496a47e6af73c0b73bd2e13c/cores/esp32/wiring_pulse.c:40: undefined reference to `xthal_get_ccount'
/home/andre/.platformio/packages/toolchain-riscv32/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/andre/.platformio/packages/framework-arduinoespressif32@src-cba3def1496a47e6af73c0b73bd2e13c/cores/esp32/wiring_pulse.c:45: undefined reference to `xthal_get_ccount'
/home/andre/.platformio/packages/toolchain-riscv32/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/andre/.platformio/packages/framework-arduinoespressif32@src-cba3def1496a47e6af73c0b73bd2e13c/cores/esp32/wiring_pulse.c:41: undefined reference to `xthal_get_ccount'
/home/andre/.platformio/packages/toolchain-riscv32/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/andre/.platformio/packages/framework-arduinoespressif32@src-cba3def1496a47e6af73c0b73bd2e13c/cores/esp32/wiring_pulse.c:42: undefined reference to `xthal_get_ccount'
/home/andre/.platformio/packages/toolchain-riscv32/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/esp_demo_c3/libFrameworkArduino.a(wiring_pulse.c.o):/home/andre/.platformio/packages/framework-arduinoespressif32@src-cba3def1496a47e6af73c0b73bd2e13c/cores/esp32/wiring_pulse.c:43: more undefined references to `xthal_get_ccount' follow
```

markusg1234, 2021-05-20:
Hi,
How do you install the tool chain?

Regards,
Markus

Tool Manager: Installing toolchain-riscv32 @ ~1.80400.0
Error: Could not find the package with 'toolchain-riscv32 @ ~1.80400.0' requirements for your system 'linux_aarch64'

misery, 2021-05-20:
It will be installed automatically. Seems linux_aarch64 is not supported at the moment. I used Linux x86_64 here.

nikhiljha, 2021-06-01:
tried this out (my esp32c-c3 arrived today) with [this config](https://github.com/esphome/feature-requests/issues/1217#issuecomment-843054221) and got a whole bunch of errors like this

```
src/esphome/core/esphal.cpp:200:10: error: 'gpio_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'status1_w1tc'; did you mean 'status_w1tc'?
     GPIO.status1_w1tc.intr_st = this->gpio_mask_;
          ^~~~~~~~~~~~
          status_w1tc
src/esphome/core/esphal.cpp: In constructor 'esphome::GPIOPin::GPIOPin(uint8_t, uint8_t, bool)':
src/esphome/core/esphal.cpp:32:55: error: 'gpio_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'out1_w1ts'; did you mean 'out_w1ts'?
           gpio_set_(pin < 32 ? &GPIO.out_w1ts : &GPIO.out1_w1ts.val),
                                                       ^~~~~~~~~
                                                       out_w1ts
src/esphome/core/esphal.cpp:33:53: error: 'gpio_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'out1_w1tc'; did you mean 'out_w1tc'?
       gpio_clear_(pin < 32 ? &GPIO.out_w1tc : &GPIO.out1_w1tc.val),
                                                     ^~~~~~~~~
                                                     out_w1tc
```

did you manage to get a binary?

stintel, 2021-06-02:
@misery I was able to build a binary after applying your PRs. Unfortunately I'm unable to flash it, as esphome-flasher also doesn't support C3 yet. I'm working on adding support here: https://github.com/stintel/esphome-flasher/tree/esp32c3.

stintel, 2021-06-02:
Flashed firmware.bin with esptool at the offset of the ota_0 partition, now getting an error about unable to set CPU clock:

```
[20:52:04]ESP-ROM:esp32c3-api1-20210207
[20:52:04]Build:Feb  7 2021
[20:52:04]rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
[20:52:04]SPIWP:0xee
[20:52:04]mode:DIO, clock div:1
[20:52:04]load:0x3fcd6100,len:0x17a8
[20:52:04]load:0x403ce000,len:0x894
[20:52:04]load:0x403d0000,len:0x2bf8
[20:52:04]entry 0x403ce000
[20:52:04]I (52) boot: ESP-IDF v4.3-beta2-2-g9a2d251912 2nd stage bootloader
[20:52:04]I (52) boot: compile time 19:26:26
[20:52:04]I (52) boot: chip revision: 3
[20:52:04]I (56) boot_comm: chip revision: 3, min. bootloader chip revision: 0
[20:52:04]I (63) boot.esp32c3: SPI Speed      : 80MHz
[20:52:04]I (67) boot.esp32c3: SPI Mode       : DIO
[20:52:04]I (72) boot.esp32c3: SPI Flash Size : 4MB
[20:52:04]I (77) boot: Enabling RNG early entropy source...
[20:52:04]I (82) boot: Partition Table:
[20:52:04]I (86) boot: ## Label            Usage          Type ST Offset   Length
[20:52:04]I (93) boot:  0 sec_cert         unknown          3f 00 0000d000 00003000
[20:52:04]I (101) boot:  1 nvs              WiFi data        01 02 00010000 00006000
[20:52:04]I (108) boot:  2 otadata          OTA data         01 00 00016000 00002000
[20:52:04]I (116) boot:  3 phy_init         RF data          01 01 00018000 00001000
[20:52:04]I (123) boot:  4 ota_0            OTA app          00 10 00020000 00190000
[20:52:04]I (131) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
[20:52:04]I (138) boot:  6 fctry            WiFi data        01 02 00340000 00006000
[20:52:04]I (146) boot:  7 coredump         Unknown data     01 03 00350000 00010000
[20:52:04]I (154) boot: End of partition table
[20:52:04]I (158) boot_comm: chip revision: 3, min. application chip revision: 0
[20:52:04]I (165) esp_image: segment 0: paddr=00020020 vaddr=3c030020 size=0cc90h ( 52368) map
[20:52:04]I (182) esp_image: segment 1: paddr=0002ccb8 vaddr=3fc8a000 size=014d8h (  5336) load
[20:52:04]I (183) esp_image: segment 2: paddr=0002e198 vaddr=40380000 size=01e80h (  7808) load
[20:52:04]I (192) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=23370h (144240) map
[20:52:05]I (221) esp_image: segment 4: paddr=00053398 vaddr=40381e80 size=08000h ( 32768) load
[20:52:05]I (228) esp_image: segment 5: paddr=0005b3a0 vaddr=50000000 size=00010h (    16) load
[20:52:05]I (231) boot: Loaded app from partition at offset 0x20000
[20:52:05]I (233) boot: Disabling RNG early entropy source...
[20:52:05][   257][E][esp32-hal-cpu.c:204] setCpuFrequencyMhz(): CPU clock could not be set to 240 MHz
```

misery, 2021-06-02:
Try to change 240 MHz to 160 MHz. C3 has 160 MHz only.

stintel, 2021-06-02:
> Try to change 240 MHz to 160 MHz. C3 has 160 MHz only.

Thanks for the suggestion. I was trying to find where this is set already, but today is the first time I look at any of the esphome related code so it's not going very fast. Is there maybe an IRC channel where I could ask things like this?

misery, 2021-06-02:
I can look into it later if I have more than a Smartphone. 😉

Esphome has a discord: https://discord.gg/KhAMKrd

stintel, 2021-06-02:
Alright found it. Your example uses board [nodemcu-32s](https://github.com/Jason2866/platform-espressif32/blob/feature/arduino-c3/boards/nodemcu-32s.json), which runs at 240MHz. The only board supported by platform-espressif32 that runs at 160MHz is bpi-bit; using that makes the CPU clock error disappear. I guess I'll be forking platform-espressif32 and add some more changes.

stintel, 2021-06-02:
So I forked platform-espressif32, and am using this config now:
```
---
esphome:
  name: esp32-c3-01
  platform: ESP32
  board: esp32-c3-dev
  platformio_options:
    board_build.mcu: esp32c3
    platform: https://github.com/stintel/platform-espressif32.git#esp32-c3
    platform_packages: framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master

wifi:
  ssid: "iot"
  password: !secret pass_wifi_iot
  power_save_mode: none
  reboot_timeout: 1h
```

This seems to trigger a watchdog timeout:
```
[21:59:47]ESP-ROM:esp32c3-api1-20210207
[21:59:47]Build:Feb  7 2021
[21:59:48]rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
[21:59:48]SPIWP:0xee
[21:59:48]mode:DIO, clock div:1
[21:59:48]load:0x3fcd6100,len:0x17a8
[21:59:48]load:0x403ce000,len:0x894
[21:59:48]load:0x403d0000,len:0x2bf8
[21:59:48]entry 0x403ce000
[21:59:48]I (52) boot: ESP-IDF v4.3-beta2-2-g9a2d251912 2nd stage bootloader
[21:59:48]I (52) boot: compile time 19:26:26
[21:59:48]I (52) boot: chip revision: 3
[21:59:48]I (56) boot_comm: chip revision: 3, min. bootloader chip revision: 0
[21:59:48]I (63) boot.esp32c3: SPI Speed      : 80MHz
[21:59:48]I (67) boot.esp32c3: SPI Mode       : DIO
[21:59:48]I (72) boot.esp32c3: SPI Flash Size : 4MB
[21:59:48]I (77) boot: Enabling RNG early entropy source...
[21:59:48]I (82) boot: Partition Table:
[21:59:48]I (86) boot: ## Label            Usage          Type ST Offset   Length
[21:59:48]I (93) boot:  0 sec_cert         unknown          3f 00 0000d000 00003000
[21:59:48]I (101) boot:  1 nvs              WiFi data        01 02 00010000 00006000
[21:59:48]I (108) boot:  2 otadata          OTA data         01 00 00016000 00002000
[21:59:48]I (116) boot:  3 phy_init         RF data          01 01 00018000 00001000
[21:59:48]I (123) boot:  4 ota_0            OTA app          00 10 00020000 00190000
[21:59:48]I (131) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
[21:59:48]I (138) boot:  6 fctry            WiFi data        01 02 00340000 00006000
[21:59:48]I (146) boot:  7 coredump         Unknown data     01 03 00350000 00010000
[21:59:48]I (154) boot: End of partition table
[21:59:48]I (158) boot_comm: chip revision: 3, min. application chip revision: 0
[21:59:48]I (165) esp_image: segment 0: paddr=00020020 vaddr=3c0a0020 size=1ada8h (109992) map
[21:59:48]I (191) esp_image: segment 1: paddr=0003add0 vaddr=3fc8cc00 size=03418h ( 13336) load
[21:59:48]I (193) esp_image: segment 2: paddr=0003e1f0 vaddr=40380000 size=01e28h (  7720) load
[21:59:48]I (198) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=99fe4h (630756) map
[21:59:48]I (303) esp_image: segment 4: paddr=000da00c vaddr=40381e28 size=0ac54h ( 44116) load
[21:59:48]I (312) esp_image: segment 5: paddr=000e4c68 vaddr=50000000 size=00010h (    16) load
[21:59:48]I (317) boot: Loaded app from partition at offset 0x20000
[21:59:48]I (317) boot: Disabling RNG early entropy source...
[21:59:53]E (5473) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[21:59:53]E (5473) task_wdt:  - IDLE (CPU 0)
[21:59:53]E (5473) task_wdt: Tasks currently running:
[21:59:53]E (5473) task_wdt: CPU 0: wifi
```

stintel, 2021-06-02:
After enabling debug, it seems the watchdog message is not fatal, my ESP32-C3 connects to my WiFi and a GPIO motion sensor seems to be working fine.

My current changes to esphome:
```diff
diff --git a/esphome/components/debug/debug_component.cpp b/esphome/components/debug/debug_component.cpp
index 4ffc034d..6c9b7ee2 100644
--- a/esphome/components/debug/debug_component.cpp
+++ b/esphome/components/debug/debug_component.cpp
@@ -5,7 +5,7 @@
 #include "esphome/core/version.h"

 #ifdef ARDUINO_ARCH_ESP32
-#include <rom/rtc.h>
+#include <esp32/rom/rtc.h>
 #endif

 namespace esphome {
diff --git a/esphome/pins.py b/esphome/pins.py
index fef77f39..b90edbcc 100644
--- a/esphome/pins.py
+++ b/esphome/pins.py
@@ -348,6 +348,7 @@ ESP32_BOARD_PINS = {
         "SS": 15,
     },
     "esp-wrover-kit": {},
+    "esp32-c3-dev": {},
     "esp32-devkitlipo": {},
     "esp32-evb": {
         "BUTTON": 34,
```

stintel, 2021-06-02:
Enabling the Home Assistant API causes a compile error:
```
src/esphome/components/api/user_services.cpp: In function 'T esphome::api::get_execute_arg_value(const esphome::api::ExecuteServiceArgument&) [with T = std::vector<int>]':
src/esphome/components/api/user_services.cpp:19:14: error: could not convert 'arg.esphome::api::ExecuteServiceArgument::int_array' from 'vector<long int>' to 'vector<int>'
   return arg.int_array;
          ~~~~^~~~~~~~~
```

stintel, 2021-06-02:
I'll double check if my changes are correct, and will submit a PR for them in the coming days, but I welcome any feedback before that.

LokiMetaSmith, 2021-06-03:
I was getting "Error: Unknown board ID" with a custom board I'm developing. This seems to be working, still testing. 

esphome:
  name: sensor
  platform: esp32
  board: custom_board
  platformio_options:
    board: esp32-c3-dev
    board_build.mcu: esp32c3
    platform: https://github.com/stintel/platform-espressif32.git#esp32-c3
    platform_packages: framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master

agners, 2021-07-18:
I successfully flashed a ESP32-C3 using the changes and yaml from this pull-request:
https://github.com/esphome/esphome/pull/2035

The GPIO fix is mostly adopted from @misery's PR.

I can successfully connect to WiFi and connect to it using Home Assistant. Haven't tested more than that currently :)

jeggleston1981, 2021-07-21:
Great, I flashed a ESP32-C3 with the new DEV ESPhome, and used 4 GPIOs as switches ok, and 1 as a GPIO binary sensor
There where a couple of issues I found:
1. The C3 has pins that are usable as GPIO that on other ESP32 are reserved for SPI memory, that the editor wouldn't let me use, I can check out the data sheet and post the usable pin if that would help?
2. BLE Improv component didn't work something about TAG, I can post log here once again if that helps. I have used BLE in arduino, I had a BLE server setup and was about to change characteristics on it.
3. Captive portal doesn't work because of ESPAsyncWebSever that doesn't work on arduino either, I assume that fix will be with ESPAsyncWebSever not here
4. Neopixel and fastLED didn't work once again can provide log if it would help

When I get some more time I've got some PWM led board that I was going to try the C3 on.

agners, 2021-07-23:
@jeggleston1981 thanks for testing and this summary.

 1. That is mostly related to non-existing board support for ESP32-C3 in `esphome/pins.py`. I am planning to add board support along with the GPIO setup for ESP32-C3-DevKitM-1.
 2. It seems we have missing `TAG` declaration in some files. This PR should fix it: https://github.com/esphome/esphome/pull/2060
 3. I did create a separate issue for the ESPAsyncWebSever crash (see https://github.com/esphome/issues/issues/2257). However, we managed to find a fix for this issue today. You can use https://github.com/agners/ESPAsyncWebServer/tree/fix-build-with-arduino-esp32-2.0.0-rc1 with all the necessary fixes included (e.g. by using `ESPAsyncWebServer-esphome=https://github.com/agners/ESPAsyncWebServer#fix-build-with-arduino-esp32-2.0.0-rc1` in `config/yourconfig/platformio.ini`).
 4. I guess those need support in upstream and then a bump to the latest version. Haven't checked the state of this yet.

stintel, 2021-07-23:
@agners thanks for picking this up. Due to travel to my home country for the first time since the start of the pandemic, and an unexpected new job, I haven't found the time to continue on this.

Seems esp32-c3 support made it in platform-espressif32 v3.3.0. Unfortunately, support for the arduino framework was left out of the board file. Ive submitted platformio/platform-espressif32#590 to fix this. For now, you can keep pulling it from my fork.

agners, 2021-07-23:
@stintel that isn't the whole change necessary, the `framework-arduinoespressif32` needs updating too, otherwise ESP32-C3 still won't build. Of course on our end we can override `framework-arduinoespressif32`, but that is not generally expected from platform.io. I've read somewhere that https://github.com/platformio/platform-espressif32 will get updated once https://github.com/espressif/arduino-esp32/ releases a stable version. So I guess we have to wait and see on that front.

Currently the platform.io repository don't carry a newer version than 1.0.6 for `framework-arduinoespressif32`. But you can reference a github repository directly too. I am using this change: https://github.com/agners/platform-espressif32/commit/2a6c8a31523a00bff76e94385acc608a4d4461f1

But there are more :snail: : It seems that the latest Arduino ESP32 v2.0.0-rc1 release uses a new toolchain. At least with the current toolchain pulled in by `platform-espressif32` I get build errors like this:

```
/root/.platformio/packages/toolchain-riscv-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: error: /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/lib/libpthread.a(pthread.c.obj): conflicting priv spec version (major/minor/revision).
/root/.platformio/packages/toolchain-riscv-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: error: /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/lib/libpthread.a(pthread.c.obj): conflicting priv spec version (major/minor/revision).
```

Using the latest `esp-2021r1` toolchain from https://github.com/espressif/crosstool-NG locally fixes that. So my config looks currently like this:

```
esphome:
  name: esp32-c3-test
  platform: ESP32
  board: esp32-c3-devkitm-1
  platformio_options:
    platform: https://github.com/agners/platform-espressif32#v3.3.0-arduino-v2.0.0-rc1
    platform_packages:
      - toolchain-riscv-esp @ file:///workspaces/esphome/toolchain-riscv-esp
    # alternatively use alpha1 which works with current toolchain
    #  - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0-alpha1
```

Where `/workspaces/esphome/toolchain-riscv-esp` basically contains what was in `~/.platformio/packages/toolchain-riscv-esp/` but with the latest crosstool-NG release from above.

That isn't taking care of ESPAsyncWebSever fixes yet, but they are under way. Things should become simpler soon :)

stintel, 2021-07-23:
You clearly know more about this than me :-) Though for me it builds fine with the yaml I posted [here](https://github.com/esphome/feature-requests/issues/1217#issuecomment-853317887). I have actually 4 C3s running with that already.

agners, 2021-07-23:
> You clearly know more about this than me :-) 

Hard earned the last two days or so :)

> Though for me it builds fine with the yaml I posted here.

You are using master. Quite possible that this did work a couple of weeks ago. But it seems with the 2.0.0-rc1 release they moved to the new compiler toolchain version.

> I have actually 4 C3s running with that already.

Cool! I have no C3 in "production" yet, but I am testing with the two dev boards I have. I am running an ESP32 with Arduino 2.0.0-rc1 to see how it behaves, looks good so far.

agners, 2021-07-25:
Actually Jason8266 brought up a good point, there is already support for the latest Arduino framework version in the `feature/arduino-idf-master` branch: It is there in the branch https://github.com/platformio/platform-espressif32/tree/feature/arduino-idf-master

This simplifies the above config to:

```yaml
esphome:
  name: esp32-c3-test
  platform: ESP32
  board: esp32-c3-devkitm-1
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32/tree/feature/arduino-idf-master
```

Still fixes for ESPAsyncWebSever and https://github.com/esphome/esphome/pull/2062 needed. But I think I'll be able to get this sorted soon.

closed 2021-07-26 by OttoWinter

digiblur, 2021-07-29:
> ```yaml
> https://github.com/platformio/platform-espressif32/tree/feature/arduino-idf-master
> ```

Fails to compile with that code.

Platform Manager: Installing git+https://github.com/platformio/platform-espressif32/tree/feature/arduino-idf-master
git version 2.20.1
Cloning into '/root/.platformio/.cache/tmp/pkg-installing-eakf123c'...
fatal: repository 'https://github.com/platformio/platform-espressif32/tree/feature/arduino-idf-master/' not found
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/platformio/platform-espressif32/tree/feature/arduino-idf-master', '/root/.platformio/.cache/tmp/pkg-installing-eakf123c']

agners, 2021-07-29:
@digiblur whoops there's a typo. Fixed it in the config above.

digiblur, 2021-07-29:
> @digiblur whoops there's a typo. Fixed it in the config above.

Getting closer ;)  

Linking .pioenvs/esp32c3ble/firmware.elf
/root/.platformio/packages/toolchain-riscv-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: error: /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/lib/libpthread.a(pthread.c.obj): conflicting priv spec version (major/minor/revision).
/root/.platformio/packages/toolchain-riscv-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: error: /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/lib/libpthread.a(pthread.c.obj): conflicting priv spec version (major/minor/revision).

agners, 2021-08-02:
@digiblur hm, it seems an old toolchain still getting used. Can you remove `/root/.platformio/packages/toolchain-riscv-esp/` and see if it compiles afterwards?

digiblur, 2021-08-02:
I did get it to compile a day or two ago but it doesn't seem to boot.  

[07:29:13]SPIWP:0xee
[07:29:13]mode:QIO, clock div:1
[07:29:13]load:0x3fcd6100,len:0x1428
[07:29:13]ets_loader.c 76 
[07:29:14]ESP-ROM:esp32c3-20200918
[07:29:14]Build:Sep 18 2020
[07:29:14]rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[07:29:14]Saved PC:0x400492ac
[07:29:14]SPIWP:0xee
[07:29:14]mode:QIO, clock div:1
[07:29:14]load:0x3fcd6100,len:0x1428
[07:29:14]ets_loader.c 76 
[07:29:15]ESP-ROM:esp32c3-20200918
[07:29:15]Build:Sep 18 2020
[07:29:15]rst:0x10 (RTCWDT_RTC_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[07:29:15]SPIWP:0xee
[07:29:15]mode:QIO, clock div:1
[07:29:15]load:0x3fcd6100,len:0x1428
[07:29:15]ets_loader.c 76 
[07:29:16]ESP-ROM:esp32c3-20200918

markusg1234, 2021-08-03:
Hi,
Is there a way I can get deep sleep working on the C3?
I get the following errors when using 

deep_sleep:
 run_duration: 60sec
 sleep_duration: 60min

Regards,
Markus

src/esphome/components/deep_sleep/deep_sleep_component.cpp: In member function 'void esphome::deep_sleep::DeepSleepComponent::begin_sleep(bool)':
src/esphome/components/deep_sleep/deep_sleep_component.cpp:78:5: error: 'esp_sleep_enable_ext0_wakeup' was not declared in this scope
     esp_sleep_enable_ext0_wakeup(gpio_num_t((*this->wakeup_pin_)->get_pin()), level);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.cpp:78:5: note: suggested alternative: 'esp_sleep_enable_wifi_wakeup'
     esp_sleep_enable_ext0_wakeup(gpio_num_t((*this->wakeup_pin_)->get_pin()), level);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
     esp_sleep_enable_wifi_wakeup
src/esphome/components/deep_sleep/deep_sleep_component.cpp:81:5: error: 'esp_sleep_enable_ext1_wakeup' was not declared in this scope
     esp_sleep_enable_ext1_wakeup(this->ext1_wakeup_->mask, this->ext1_wakeup_->wakeup_mode);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.cpp:81:5: note: suggested alternative: 'esp_sleep_enable_wifi_wakeup'
     esp_sleep_enable_ext1_wakeup(this->ext1_wakeup_->mask, this->ext1_wakeup_->wakeup_mode);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
     esp_sleep_enable_wifi_wakeup
*** [/data/esp32-c/.pioenvs/esp32-c/src/esphome/components/deep_sleep/deep_sleep_component.cpp.o] Error 1
Compiling /data/esp32-c/.pioenvs/esp32-c/src/esphome/core/application.cpp.o
========================== [FAILED] Took 2.55 seconds ==========================

agners, 2021-08-06:
@markusg1234 haven't tested sleeping modes so far, interesting find. Can you open a [new issue](https://github.com/esphome/issues/issues/new/choose)?

agners, 2021-08-06:
In general, as basic support for ESP32-C3 is in, I think it make sense to track further issues/missing pieces via issue tracker. Feel free to tag me in those issues.

agners, 2021-09-12:
@digiblur  I do get this now too, try this option:

```
  platformio_options:
    board_build.flash_mode: dio
```

I am pretty sure that Platform.IO's board configuration used QIO from the beginning, and it did work at one point. It seems that some change in Arduino/Platform.IO in late July/early August  broke QIO support.