esphome/feature-requests #1221: adding outdoor temperature workaround for midea_ac component

dszabolcs, 2021-05-11:
Many midea climate does not support to measuring outdoor temperature when the AC is turned off. In this case, the outdoor temperature shows 102.5 C which occurs issues on the Home Assistant level (for example some automations are activated above a certain outdoor temperature)

My suggestion is to create a new parameter like 'outdoor_temperature_off_workaround' where the user can decide that how much temperature is reported when the outdoor temperature is bigger than 100 C or equal with 102 or 102.5 (depends on AC). And it is possible to add an option that if the user wants to report the last valid outdoor temperature when the AC is off.

In this case the parameter yaml structure look like this:

outdoor_temperature_off_workaround:
--- off_temperature: 20 // optional, mutual exclusive with last_valid_state, if given than the esp reports always 20 C when the AC is turned off and reports 102.5C
--- last_valid_state: true/false // optional, mutual exclusive with off_temperature, if given the esp reports always the last known valid outdoor temperature

dszabolcs, 2021-05-11:
I can also implement it, if you don't have any effort to do this.

nagyrobi, 2022-07-07:
Please do so and submit a PR or publish it as an external component for others.