esphome/feature-requests #1223: Add Support for Panasonic H & J Series Aquarea air-water heat pumps

pasarn, 2021-05-12:
**Describe the problem you have/What new integration you would like**

The Panasonic H & J Series Aquarea air-water heat pump had same sensors and settings. This settings and sensors are visible over UART. The communication protocol is reverse engineered by Egyras (https://github.com/Egyras/HeishaMon/blob/master/ProtocolByteDecrypt.md)

**Please describe your use case for this integration and alternatives you've tried:**

I wand to monitor and control the head pump with Home Assistant. Egyras has an nice solution for this. But I think it is better to use esphome. One ESP ecosystem for many use cases.

**Additional context**

Many information's are in Egyras repository (https://github.com/Egyras/HeishaMon)
It exist an ready to use PCB to connect the esp to the head pump (https://www.tindie.com/products/thehognl/heishamon-communication-pcb/)

nagyrobi, 2022-07-08:
Implemented in https://esphome.io/components/climate/climate_ir.html, multiple Panasonic protocols.

pasarn, 2022-07-08:
The Aquarea dosn't communicate over IR. It is an UART Interface. Please reopen the request.