esphome/feature-requests #1224: Analog Output of ESP32 Values for Soil Moisture Measurements

evilsidekick293, 2021-05-17:
**Describe the problem you have/What new integration you would like**

I would like see the analog value of a ESP32 touch pin (much like an adc sensor).

**Please describe your use case for this integration and alternatives you've tried:**

I would love to measure soil moisture in my plants soil.

**Additional context**

I would like to build something like a Xiaomi Flora.

torfbolt, 2021-09-05:
Second this, using ESP32 touch as a non-binary sensor would be cool for various capacitive sensing applications, e.g. rain sensor or similar.

nagyrobi, 2022-07-07:
See: https://esphome.io/components/sensor/adc.html
Can touch pins be configured there?