esphome/feature-requests #1226: Support for PZEM-017/003 Energy Measurement

speakerboar, 2021-05-18:
**Describe the problem you have/What new integration you would like**

PZEM 017/003 Support for Energy Sensor
**Please describe your use case for this integration and alternatives you've tried:**

The PZEM-017 and PZEM-003 measures DC volts , Power , Ampere and Energy but there is no support for Energy in esphome
**Additional context**

This feature is implemented in pzemac and pzem004t but those are AC

andreashergert1984, 2021-08-14:
> **Describe the problem you have/What new integration you would like**
> 
> PZEM 017/003 Support for Energy Sensor - [pzemdc](https://esphome.io/components/sensor/pzemdc.html)
> **Please describe your use case for this integration and alternatives you've tried:**
> 
> The PZEM-017 and PZEM-003 measures DC volts , Power , Ampere and Energy but there is no support for Energy in esphome
> **Additional context**
> 
> This feature is implemented in [pzemac](https://esphome.io/components/sensor/pzemac.html) and [pzem004t](https://esphome.io/components/sensor/pzem004t.html) but those are AC
might you please check if https://esphome.io/components/sensor/pzemdc.html is working for you? since the PZEM-017 is listed as supported there.

thx
Andi

speakerboar, 2021-08-14:
This feature request is to add energy (kWh). The PZEM-017 is supported in esphome but energy (kWh) specifically is missing.

nagyrobi, 2022-07-06:
You can configure the `total_daily_energy` separately for sensor which don't have it built-in:
https://esphome.io/components/sensor/total_daily_energy.html

closed 2022-07-06 by nagyrobi

speakerboar, 2022-07-06:
The device already measure kWh just this component not able to read it. 
total daily energy is not really a solution as its not as accurate and it resets whenever the ESP restarts , utility meter not good as well as I don't want it depended on home assistant

I was able to get the energy reading use the [modbus_component](https://github.com/esphome/feature-requests/issues/1245#issuecomment-934723925)

but it would be nice if the pzemdc component include it

andreashergert1984, 2022-07-06:
> The device already measure kWh just this component not able to read it. total daily energy is not really a solution as its not as accurate and it resets whenever the ESP restarts , utility meter not [...]

i did it already in a branch.. i will make it nice and make a pull request.

for everybody who wants already use it:

external_components:
  - source: github://andreashergert1984/esphome@add_energy_to_pzemdc
    refresh: 0s
    components: [pzemdc]

  - platform: pzemdc
    address: 2
    current:
      name: "MPPT Charger Adrress 2 Current"
      filters:
        multiply: 0.5
    voltage:
      name: "MPPT Charger Address 2 Voltage"
    power:
      name: "MPPT Charger Address 2 Power"
      filters:
        multiply: 0.5
    energy:
      name: "MPPT Charger Address 2 Energy"
      filters:
        multiply: 0.5