esphome/feature-requests #1228: Force save boot restart switch.

jaddel, 2021-05-20:
**Describe the problem you have/What new integration you would like**

A new component "force save boot restart switch"

**Please describe your use case for this integration and alternatives you've tried:**

In some configurations i manged to get the WiFi very unstabile. Then the web interface is still accessible for short periods of time but the connection is to week to finish an ota update.  

**Additional context**
This happend for me while i was playing around with the ble tracker component on esp32.

TheJulianJES, 2021-06-04:
I'd love a simple "Reboot" button in the UI for each esphome node. (Can't seem to trigger a full WiFi scan otherwise)

jaddel, 2021-06-04:
it's not in the UI but this might help you:
https://esphome.io/components/switch/restart.html

nagyrobi, 2022-07-06:
You need to put this in the config: https://esphome.io/components/button/safe_mode.html
It will appear both in HA and web UI.

closed 2022-07-06 by nagyrobi