esphome/feature-requests #1229: Detail the unit of BH1750 measurement_duration

wernight, 2021-05-20:
**Describe the problem you have/What new integration you would like**
https://github.com/esphome/esphome-docs/blob/current/components/sensor/bh1750.rst has a `measurement_duration` (previously `measurement_time`) but I don't know the unit of this parameter just that it's an int less than 254. For example increasing this to 254 would it mean that I can only get an updated value after 254 seconds? milliseconds?

**Please describe your use case for this integration and alternatives you've tried:**
I looked at https://github.com/esphome/esphome/blob/dev/esphome/components/bh1750/bh1750.cpp but didn't understand the unit. It seems to be handled by BH1750 internally. From the [datasheet](https://www.mouser.com/datasheet/2/348/bh1750fvi-e-186247.pdf) section "Adjust measurement result for influence of optical window" it seems that measurement_duration is in millis. If confirmed I'm add that to the documentation. I'd even consider changing the default as 250 ms is fast for more ESPHome applications.

**Additional context**
None

yuvalabou, 2021-05-24:
That is actually very confusing and even the datasheet is foggy about it. I would assume that it's in millis - but even so why does it matter? What are the suggestion?  In the datasheet the default is also says 69 for MTreg

OttoWinter, 2022-02-19:
Closing per https://github.com/esphome/esphome/pull/3214 - the best mtreg value is dynamically calculated, and the user doesn't need to choose these anymore

closed 2022-02-19 by OttoWinter