esphome/feature-requests #1230: Support Yeelight Wireless Switch S1 - YLAI003

andrewjswan, 2021-05-20:
**Describe the problem you have/What new integration you would like**
**Yeelight Wireless Switch S1 - YLAI003** - https://aliexpress.ru/item/1005001552119268.html
The button has three actions, **press, double press, long press**.
There is also a battery charge value.
When i add it to MiHome, a bindkey was generated.

**Please describe your use case for this integration and alternatives you've tried:**

Manage smart home devices, transfer clicks to the HA.

**Additional context**

Already support by [BLE-monitor](https://github.com/custom-components/ble_monitor/pull/274)

andrewjswan, 2021-05-27:
More info: https://github.com/custom-components/ble_monitor/issues/268

dentra, 2021-09-24:
You could use it as custom component: https://github.com/dentra/esphome-components/tree/master/components/miot_ylyk01yl

andrewjswan, 2021-09-25:
Thanks, work!

closed 2022-07-06 by nagyrobi