esphome/feature-requests #1231: Cover on_stop triggers

muxa, 2021-05-22:
**Describe the problem you have/What new integration you would like**
I'd like to be able to trigger some action when a cover stops (e.g. it was closing or opening and then it stopped, either fully closed/opened or in between).

**Please describe your use case for this integration and alternatives you've tried:**
I have a time-based cover and 2 template switches that control opening and closing (switch is on - cover is moving, switch is off - cover stops). I want to be able to turn off those switches automatically when cover has stopped moving. The problem is that I don't know when is stops.

**Additional context**
Here's a snippet of my yaml file:

```yaml
switch:
  - platform: template
    id: btn_open
    internal: true
    turn_on_action:
    - switch.turn_on: pin_open
    - delay: 300ms
    - switch.turn_off: pin_open

  - platform: template
    id: btn_stop
    internal: true
    turn_on_action:
    - switch.turn_on: pin_stop
    - delay: 300ms
    - switch.turn_off: pin_stop

  - platform: template
    id: btn_close
    internal: true
    turn_on_action:
    - switch.turn_on: pin_close
    - delay: 300ms
    - switch.turn_off: pin_close   

  - platform: template
    id: switch_open
    name: "${device_name} Open"
    lambda: |-
        return id(direction) == 1;
    turn_on_action:
    - lambda: |-
        id(direction) = 1;
    - cover.open: louvres
    turn_off_action:
    - lambda: |-
        id(direction) = 0;
    - cover.stop: louvres

  - platform: template
    id: switch_close
    name: "${device_name} Close"
    lambda: |-
        return id(direction) == -1;
    turn_on_action:
    - lambda: |-
        id(direction) = -1;
    - cover.close: louvres
    turn_off_action:
    - lambda: |-
        id(direction) = 0;
    - cover.stop: louvres
    
cover:
  - platform: time_based
    name: "${device_name}"
    id: louvres

    device_class: shutter

    open_action:
      - switch.turn_on: btn_open
    open_duration: 11s

    close_action:
      - switch.turn_on: btn_close
    close_duration: 10s

    stop_action:
      - switch.turn_on: btn_stop
```

For brevity I have omitted GPIO config for `pin_open`, `pin_stop`, `pin_close`  that actually controls the cover.

muxa, 2021-05-22:
I guess I could just add a delay to `switch_open` and `switch_close` actions, but it would be nice to drive the state of the switches from the cover, e.g:

```yaml
cover:
  - platform: time_based
    name: "${device_name}"
    id: louvres

    device_class: shutter

    open_action:
      - switch.turn_on: btn_open
    open_duration: 11s

    close_action:
      - switch.turn_on: btn_close
    close_duration: 10s

    stop_action:
      - switch.turn_on: btn_stop
   
   on_stop 
      - switch.turn_off: switch_open   
      - switch.turn_off: switch_close   
```

muxa, 2021-05-22:
I ended up adding delays that match by timed-based cover settings:

```yaml
  - platform: template
    id: switch_open
    name: "${device_name} Open"
    lambda: |-
        return id(direction) == 1;
    turn_on_action:
    - lambda: |-
        id(direction) = 1;
    - cover.open: louvres
    - delay: 11s
    - lambda: |-
        id(direction) = 0;
    turn_off_action:
    - lambda: |-
        id(direction) = 0;
    - cover.stop: louvres

  - platform: template
    id: switch_close
    name: "${device_name} Close"
    lambda: |-
        return id(direction) == -1;
    turn_on_action:
    - lambda: |-
        id(direction) = -1;
    - cover.close: louvres
    - delay: 10s
    - lambda: |-
        id(direction) = 0;
    turn_off_action:
    - lambda: |-
        id(direction) = 0;
    - cover.stop: louvres
```

glmnet, 2021-05-22:
https://esphome.io/api/classesphome_1_1cover_1_1_cover.html#a15a7743cf61a825044b8a0f2c0b9531e

This? For your template binary sensors

glmnet, 2021-05-22:
Mmm it doesn't make much sense for me as switches, but you can use that  I guess

muxa, 2021-05-22:
Thanks @glmnet, I missed that somehow.

closed 2021-05-22 by muxa