esphome/feature-requests #1235: W5500 spi ethernet

dyrvigk, 2021-05-24:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

I would like to get support for w5500 lan spi interface and using ethernet2 libary  the reason is that the supported lan8720 only works with esp32 and it is hardware wise more dificult to get to work  
**Additional context**

evi-snowm, 2021-06-24:
The possability of supporting W5500 as Ethernet option would also be relevant for people interested in using the M5 Atom series of ESP32s. They have a PoE enabled ethernet accessory ( info & specs: https://docs.m5stack.com/en/atom/atom_poe ); this would eliminate power supplies and make having a collection of these a lot easier. Perfectly willing to buy a few to use for testing.

barnabyjones, 2021-11-30:
I was just looking at that M5 Atom POE device too. What is involved in adding support for the w5500 Ethernet chipset?

nagyrobi, 2022-03-21:
Another nice ESP32 product with W5500:
https://kmpelectronics.eu/products/prodino-esp32-ethernet-v1/
Specs and module datasheets at the bottom of the page.

nagyrobi, 2022-06-14:
USR-ES1 also W5500:
[image]

dyrvigk, 2022-06-14:
i think what vis needed is to make a custom component since it is supported in arduino and libaryies are there ethernet2 and spi is already supported but it seems that Otto Winter dont think there is enough interest for this i myself have used it before in arduino and hardware vise it is much easyier to work with than lan8720.  specialy hardware vise i am now using lan 8720 but is is a pain in the butt to make pcb with because of the need for diferential pair routing due to 50mhz clock

jbabio, 2022-06-17:
Another bunch of ESP32 PLCs with W5500 pay
https://www.industrialshields.com/industrial-esp32-plc-products-family-ideal-for-iot-solutions

jburget, 2022-06-28:
Hey, I would also hope for this to get it supported, industrial shields are amazing products

philipptrenz, 2022-06-30:
Just took a look at the _Ethernet_ component ([here](https://github.com/esphome/esphome/blob/dev/esphome/components/ethernet/ethernet_component.cpp)) and support for `lan8720` as well as `phy_tlk110` is provided directly be ESP-IDF. Since v4.3 (or even earlier) there also seem to be support for W5500 in ESP-IDF (see [here](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32/api-reference/kconfig.html?highlight=w5500#config-eth-spi-ethernet-w5500)). 
Maybe it is not that hard to add native support for W5500 to the _Ethernet_ component?

nagyrobi, 2022-06-30:
Hmm...
One extra step is definitely needed: https://github.com/esphome/feature-requests/issues/1683
Because with ESPHome set to `type: esp-idf` won't do any Ethernet at all...

dyrvigk, 2022-06-30:
w5500 works fine in arduino  using ethernet or ethernet2  libary   Khoi
Hoang
has made some nice webserver examples look at his github

philipptrenz, 2022-07-07:
Managed to get a minimal working example using the Arduino framework and the lwip stack of the WiFi library. 

```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <ETH.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include "driver/spi_master.h"
#include <SPI.h>

/* 
 * GPIO config for W5500: 
 * 
 * MISO: GPIO19
 * MOSI: GPIO23
 * SCLK: GPIO18
 * SCS:  GPIO5
 * INT:  GPIO22
 */

AsyncWebServer server(80);

bool setupW5500() {

  WiFi.begin();
  tcpip_adapter_set_default_eth_handlers();

  // Initialize TCP/IP network interface
  ESP_ERROR_CHECK(esp_netif_init());
  esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
  esp_netif_t *eth_netif = esp_netif_new(&cfg);

  esp_eth_mac_t *eth_mac = NULL;
  esp_eth_phy_t *eth_phy = NULL;
  gpio_install_isr_service(0);

  spi_bus_config_t buscfg = {
    .mosi_io_num = 23,  // MOSI
    .miso_io_num = 19,  // MISO
    .sclk_io_num = 18,  // SCLK
    .quadwp_io_num = -1,
    .quadhd_io_num = -1,
  };
  ESP_ERROR_CHECK(spi_bus_initialize(SPI3_HOST, &buscfg, 1));

  spi_device_handle_t spi_handle = NULL;
  spi_device_interface_config_t devcfg = {
      .command_bits = 16,   // Address phase in W5500 SPI frame
      .address_bits = 8,    // Control phase in W5500 SPI frame
      .mode = 0,
      .clock_speed_hz = 12 * 1000 * 1000,
      .spics_io_num = 5,    // SCS
      .queue_size = 20
  };
  ESP_ERROR_CHECK(spi_bus_add_device(SPI3_HOST, &devcfg, &spi_handle));

  /* W5500 ethernet driver uses spi driver */
  eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);
  w5500_config.int_gpio_num = 22;   // INT

  eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
  eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
  phy_config.reset_gpio_num = -1;

  eth_mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config);
  if(eth_mac == NULL){
    log_e("esp_eth_mac_new_esp32 failed");
    return false;
  }

  eth_phy = esp_eth_phy_new_w5500(&phy_config);
  if(eth_phy == NULL){
    log_e("esp_eth_phy_new failed");
    return false;
  }
    
  esp_eth_config_t eth_config = ETH_DEFAULT_CONFIG(eth_mac, eth_phy);
  esp_eth_handle_t eth_handle = NULL;
  ESP_ERROR_CHECK(esp_eth_driver_install(&eth_config, &eth_handle));

  uint8_t macArr[] = { 0x02, 0x00, 0x00, 0x12, 0x34, 0x56 };
  ESP_ERROR_CHECK(esp_eth_ioctl(eth_handle, ETH_CMD_S_MAC_ADDR, macArr));

  /* Attach Ethernet driver to TCP/IP stack */
  ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));

  /* Start Ethernet driver state machine */
  ESP_ERROR_CHECK(esp_eth_start(eth_handle));

  return true;
}

void setupWebserver() {
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
      request->send(200, "text/plain", "Hello world");
  });
  server.onNotFound([](AsyncWebServerRequest *request){ 
    request->send(404, "text/plain", "Not found");
  });
  server.begin();
}

void setup() {
  Serial.begin(115200);
  setupW5500();
  setupWebserver();
}

void loop() {}
```

From here, it might be manageable to integrate, right? A short assessment from @OttoWinter or maybe @jesserockz would be very helpful!

nagyrobi, 2022-10-27:
Unfortunately, no

nagyrobi, 2022-12-07:
https://github.com/esphome/esphome/pull/3564 upgrades ethernet for ESPHome to include 2 more drivers and esp-idf support added in https://github.com/esphome/esphome/pull/3565.

This should make it possible to add further Ethernet drivers more easily.

alve89, 2022-12-24:
Since using the W5500 is so pretty easy on Arduino, I just stumbled over this FR. I'd also appreciate an integration of W5500 support in ESPHome! 😊

Thomas7173, 2023-02-06:
I would like to use ESP32 plus W5500 for energy monitoring via S= inside a metallic switchboard. so WIFI is no option. Is any work going on in implementing the W5500 board?

JeroenVanOort, 2023-02-13:
I have just made a draft PR to add support for the W5500 board. I have not been able to test this on an actual device yet, but anyone is welcome to give it a try!

mhendriks, 2023-03-12:
Thanks!!! Works great with some changes for the ESP32C3 with W5500 25 MHz clock.

`#include <Arduino.h>
#include <WiFi.h>
#include <ETH.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include "driver/spi_master.h"
#include <SPI.h>

AsyncWebServer server(80);

bool setupW5500() {

  WiFi.begin();
  tcpip_adapter_set_default_eth_handlers();

  // Initialize TCP/IP network interface
  ESP_ERROR_CHECK(esp_netif_init());
  esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
  esp_netif_t *eth_netif = esp_netif_new(&cfg);

  esp_eth_mac_t *eth_mac = NULL;
  esp_eth_phy_t *eth_phy = NULL;
  gpio_install_isr_service(0);

  spi_bus_config_t buscfg = {
    .mosi_io_num = 6,  // MOSI
    .miso_io_num = 5,  // MISO
    .sclk_io_num = 4,  // SCLK
    .quadwp_io_num = -1,
    .quadhd_io_num = -1,
  };
//SPI_DMA_CH_AUTO for the ESP32C3
//SPI2_HOST ... the only one left for the ESP32C3
  ESP_ERROR_CHECK(spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO));

  spi_device_handle_t spi_handle = NULL;
  spi_device_interface_config_t devcfg = {
      .command_bits = 16,   // Address phase in W5500 SPI frame
      .address_bits = 8,    // Control phase in W5500 SPI frame
      .mode = 0,
      .clock_speed_hz = 25 * 1000 * 1000,
      .spics_io_num = 10,    // SCS
      .queue_size = 20
  };
//SPI2_HOST ... the only one left for the ESP32C3
  ESP_ERROR_CHECK(spi_bus_add_device(SPI2_HOST, &devcfg, &spi_handle));

  /* W5500 ethernet driver uses spi driver */
  eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);
  w5500_config.int_gpio_num = 1;   // INT

  eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
  eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
  phy_config.reset_gpio_num = -1;

  eth_mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config);
  if(eth_mac == NULL){
    log_e("esp_eth_mac_new_esp32 failed");
    return false;
  }

  eth_phy = esp_eth_phy_new_w5500(&phy_config);
  if(eth_phy == NULL){
    log_e("esp_eth_phy_new failed");
    return false;
  }
    
  esp_eth_config_t eth_config = ETH_DEFAULT_CONFIG(eth_mac, eth_phy);
  esp_eth_handle_t eth_handle = NULL;
  ESP_ERROR_CHECK(esp_eth_driver_install(&eth_config, &eth_handle));

  uint8_t macArr[] = { 0x02, 0x00, 0x00, 0x12, 0x34, 0x56 };
  ESP_ERROR_CHECK(esp_eth_ioctl(eth_handle, ETH_CMD_S_MAC_ADDR, macArr));

  /* Attach Ethernet driver to TCP/IP stack */
  ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));

  /* Start Ethernet driver state machine */
  ESP_ERROR_CHECK(esp_eth_start(eth_handle));

  return true;
}

void setupWebserver() {
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
      request->send(200, "text/plain", "Hello world W5500 test");
  });
  server.onNotFound([](AsyncWebServerRequest *request){ 
    request->send(404, "text/plain", "Not found");
  });
  server.begin();
}

void setup() {
  Serial.begin(115200);
  USBSerial.begin(115200); //initialise CDC output
  setupW5500();
  setupWebserver();
//  USBSerial.print(F("ETH IP : ")); //CDC output
//  USBSerial.println(ETH.localIP().toString());
//  USBSerial.print(F("ETH MAC: ")); //CDC output
//  USBSerial.println(ETH.macAddress());
}

void loop() {}
`

DjordjeMandic, 2023-03-19:
Basically if this gets added we can use esp32c3 for eth-ble bridge?

NajibOladosu, 2023-04-07:
Sorry, I'm new to this but i need to know if theUSR-ES1 W5500 can be used to power a pyboard directly or is some device needed. If I need a device, can I get your recommendations?

yellobyte, 2023-06-17:
Another ESP32-Ethernet board variant hosting a W5500: https://github.com/yellobyte/ESP32-DevBoards-Getting-Started/tree/main/boards/YB-ESP32-S3-ETH(YelloByte) ...and waiting to get native support.
As indicated under https://github.com/platformio/platform-espressif32/issues/1123 this is not targeted for Arduino Cores < 3.0.0.  However, Ethernet(2) lib is sufficient for most requirements though.

vitoraugustodesm, 2023-07-21:
I bought this w5500 shield and was hoping w5500 could be natively supported on esphome. https://aliexpress.com/item/1005003589665341.html

fightforlife, 2023-10-28:
See this PR discussion for working ethernet component on ESP32 Arduino and ESP-IDF.
https://github.com/esphome/esphome/pull/4424

cybershoe, 2023-11-15:
I'd love to see this extended to include the rp2040, not just ESP-IDF. The PoE Featherwing along with the RP2040-shim makes for a very compact platform for wired PoE sensors.

pixelwave, 2023-12-27:
Waiting eagerly for native W5500 support to use the **Lilygo T-ETH-Lite S3** with ESPHome!

linkedupbits, 2024-01-13:
I have tested this with
- https://github.com/JeroenVanOort/esphome (using the eth-w5500 branch)
- lily-go t-relay
- lily-go t-relay W5500 shield

It works, but needs to have a slower clock speed .
This is my YAML (referencing the external components):
``` yaml
esphome:
  name: t-relay
  friendly_name: t-relay

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    

external_components:
  - source:
      type: git
      url: https://github.com/JeroenVanOort/esphome/
      ref: eth-w5500
    components: [ ethernet,network,api,esp32]

# Enable logging
# Logger must be at least debug (default)
logger:
  level: DEBUG

# Enable Home Assistant API
# Enable Home Assistant API
api:
  encryption: 
    key: !secret grandviewcontrol_encryption_key

ota:

ethernet:
  type: W5500
  clk_pin: GPIO22
  mosi_pin: GPIO26
  miso_pin: GPIO13
  cs_pin: GPIO27
  interrupt_pin: GPIO36
  reset_pin: GPIO23
  clock_speed: 16Mhz #works
#  clock_speed: 32Mhz #doesn't
```

closed 2024-03-01 by jesserockz

pixelwave, 2024-03-12:
Trying to access the ESP32 connected via POE/Ethernet results now in the message:

`
INFO ESPHome 2024.2.2
INFO Reading configuration /config/woesp32-s3-7p-eth.yaml...
ERROR Unable to import component ethernet:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/.esphome/external_components/2f02e929/esphome/components/ethernet/__init__.py", line 11, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_SPI' from 'esphome.const' (/esphome/esphome/const.py)
Failed config

api: [source /config/woesp32-s3-7p-eth.yaml:17]
  
  Component api requires component network.
  encryption: 
    key: 2RUdO7HGxXxo0WOYrW6MXLJ3axAMqkLLGCSHsPjszVk=
ota: [source /config/woesp32-s3-7p-eth.yaml:21]
  
  Component ota requires component network.
  password: 0a20e03c82db327d1524d8c2b8c8d555
ethernet: [source /config/woesp32-s3-7p-eth.yaml:31]
  
  Component not found: ethernet.
  type: W5500
  clk_pin: GPIO10
  mosi_pin: GPIO12
  miso_pin: GPIO11
  cs_pin: GPIO09
  interrupt_pin: GPIO13
  reset_pin: GPIO14
  clock_speed: 30Mhz
web_server_base: None
  {}
Component web_server_base requires component network
`

dyrvigk, 2024-03-12:
> Ah ok thanks for clarification!
>
> If it is already in the dev then I guess it will be soon also in the
> release version if testing goes well .. right?

I think so

Kenn

archef2000, 2024-03-22:
I got the ethernet W5500 component working with:
```yaml
ethernet:
  domain: .fritz.box
  type: w5500
  clk_pin: GPIO15 # SCK
  mosi_pin: GPIO14 # MO
  cs_pin: GPIO13 # CS
  miso_pin: GPIO12 # M
  clock_speed: 20Mhz
```
But I can't get any higher clock speed than 20Mhz.
Do I need the interrupt/reset pin to get anything higher?

dyrvigk, 2024-03-23:
Yes without interupt it is very slow