esphome/feature-requests #1236: Ability to turn wifi on and off

yuvalabou, 2021-05-24:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

Right now the wifi connection is on unless in deep sleep, but in deep sleep mode we cannot take any measurements from sensor.
My proposal is to be able to enable or disable the wifi module on the chip but still run the code, and enable the wifi to transmit only desired changes - hence prolonging battery life.

**Additional context**

rradar, 2021-05-24:
this is about th esp32 ulp coprocessor? If yes this might be duplicate: https://github.com/esphome/feature-requests/issues/826

yuvalabou, 2021-05-24:
Not realy.. its about the option to disable wifi unless I want to broadcast changes, but still use the node to take measurements or standalone automations offline.

For example - https://circuits4you.com/2019/01/08/esp8266-turn-off-wifi-save-power/

glmnet, 2021-05-24:
There is an open PR about this

yuvalabou, 2021-05-24:
Thanks #1249

closed 2021-05-24 by yuvalabou