esphome/feature-requests #1239: Add GIT support via UI for configuration management

XtremeOwnageDotCom, 2021-05-29:
**Describe the problem you have/What new integration you would like**
The ability to specify a upstream git repository for managing configuration versioning.

**Please describe your use case for this integration and alternatives you've tried:**
Currently, the workaround is to manually initialize a git repository in the esphome data directory, and to either manually trigger a git add/git push after updates are performed, or to setup a cron job to invoke a script to check in changes to a git repository.

Ideally- this proposed feature, would enable some basic git integration, to automatically push changes to a upstream git repository. As a future goal, the ability to see previous version directly via the esphome UI would be nice too.

**Additional context**

gregmac, 2022-03-16:
I'm a bit surprised more people aren't asking for this. This lets you freely make experimental changes to a working setup, and then "go back" to an earlier version in case it doesn't work out.

A viable MVP for this feature could be:

* Initialize a local repository
* Allow specifying a remote push-only repository
* Commit every time something is saved
* Push after every commit

This would allow viewing history on the remote repository and at least getting an earlier verison.

Nice to have or future features (in no particular order):

1. Allow specifying the commit message
1. Commit manually instead of at every save
1. Pull changes as long as there are no conflicts
1. When a firmware is installed, tag it with the install datetime
1. View log and revert to older versions
2. Individual repositories for each device
3. Allow adding a device by cloning a remote repository

nagyrobi, 2022-07-07:
Think off the fact that the concept is to use the entire system offline, most people use the Dashboard as Home Assistant add-on...

lacrimal, 2024-10-01:
Maybe this:

sh script inside config directory `auto_git_push.sh`
```
#!/bin/bash

shopt -u extglob
# Directory containing your ESPHome yaml files
WATCH_DIR="/home/me/esphome/config"
SSH_KEY_PATH="/home/me/esphome/config/id_ed25519"
BRANCH="master"  # Change if needed

# Store the GIT_SSH_COMMAND for reuse
GIT_CMD="GIT_SSH_COMMAND='ssh -i $SSH_KEY_PATH' git -C $WATCH_DIR"

# Watch for changes in .yaml files
inotifywait -m -e modify,create,delete,move --format '%w%f' --exclude '(\.git|\.esphome|trash|secrets\.yaml|id_ed25519|id_ed25519\.pub|auto_notify_push\.sh)' "$WATCH_DIR" |
while read -r FILE
do
  if [[ "$FILE" == *.@(yaml|csv) ]]; then
    # Add the modified file
    eval "$GIT_CMD add \"$FILE\""

    # Get the number of lines changed for this file
    LINES_CHANGED=$(eval "$SSH_CMD $GIT_CMD diff --cached --stat \"$FILE\" | awk '{print \$1}'")

    # Personalized commit message
    COMMIT_MSG="Updated $LINES_CHANGED lines in $(basename "$FILE")"

    # Commit and push changes
    eval "$GIT_CMD commit -m \"$COMMIT_MSG\"" && \
    eval "$GIT_CMD push origin \"$BRANCH\""
  fi
done
```
service `esphome-auto-git-push.service`
```
[Unit]
Description=Automatic Git Add, Commit, and Push for YAML files in ESPHome Directory
After=network.target

[Service]
ExecStart=/home/me/esphome/config/auto_git_push.sh
WorkingDirectory=/home/me/esphome/config
Restart=always
User=root
[Install]
WantedBy=multi-user.target
```
`.gitignore`
```
# Gitignore settings for ESPHome
# This is an example and may include too much for your use-case.
# You can modify this file to suit your needs.
/.esphome/
/.git/
/secrets.yaml
/trash/
id_ed25519
id_ed25519.pub
```
you need to:
- add inotifywait to your system
- initialize git repo localy in config folder as usual
- create key pair in same directory like mine id_ed25519 and id_ed25519.pub
- create remote repo as you like (mine is gitlab so i needed to add above id_ed25519.pub to repo keys and give privileges)

So anytime when you make save in esphome it will automatically add,  commit to local repo and push to remote.
As you see - only yaml and csv extension go automatically, rest you can do manually if you like.

That make automate half of your concern, all other you need to do manually :-)

XtremeOwnageDotCom, 2024-10-02:
> That make automate half of your concern, all other you need to do manually :-)

Oh, I already have that part scripted/automated. But- the basic functionality, would be quite handy for the product itself.

Pretty simple one too.