esphome/feature-requests #1240: More general custom RC protocol for remote transmitter

omegacore, 2021-05-29:
**Describe the problem you have/What new integration you would like**

Remote transmitter rc protocol should allow more than 2 values for sync and other field lists.

**Please describe your use case for this integration and alternatives you've tried:**

I am trying to integrate a BF-301 controlled shade system and it requires a protocol described here:
[BF-301 protocol](https://github.com/akirjavainen/markisol/blob/master/Markisol.ino)

I could use the raw transmit option, but this is tedious and the custom RC protocol would be sufficient and more efficient if I could better customize the sync, one, and zero fields.

**Additional context**

lanrat, 2021-05-30:
I'd also like this (for various other IR protocols).

However in the meantime, I've had really good luck taking the RAW IR signals, running them through [IrScrutinizer](https://appimage.github.io/IrScrutinizer/) to clean up the IR code, and then using the cleaned values to transmit from ESPHome.

It adds a few extra steps, but I've found it greatly increases reliability when capturing & retransmitting raw values.

omegacore, 2021-08-25:
Oh man, I am going to try IrScrutinizer for sure. Very cool. I was using a usb-oscope and esphome's capture system. I think I was struggling with a faulty esp32 board, among other things. RF signals seem to be pretty noisy. The oscope capture was OK, but very difficult to parse. This tool should help.

I might try to take a crack at opening a PR to do what I have requested here. Seems like a good first contribution.