esphome/feature-requests #1242: External components directory

oxan, 2021-05-30:
This isn't really a feature request, but I suppose this is the best place to start this discussion. Hopefully it'll go somewhere.

ESPHome v1.18 introduced [external components](https://esphome.io/components/external_components.html). I suspect that over time more and more devices and functionality will be available as an external component instead of a built-in one. And that's (in my opinion) a good thing, as it allows the maintenance load to be shared among more people and the release cycles to be decoupled.

However, this comes with the downside that while internal components are easily discoverable through esphome.io, external components are a lot harder to find: it currently more or less boils down to stumbling upon a GitHub repository that may or may not be easily findable through Google. I think it'd be great if there was some sort of external component directory for ESPHome, comparable to what HACS is for Home Assistant components.

I see a few options:
* The simplest option is to appropriate a GitHub topic for this (e.g. [esphome-component](https://github.com/topics/esphome-component)), but that doesn't give a great UI.
* It's integrated in the esphome.io site. This is definitely the best option for discoverability, but it carries an implicit endorsement by the ESPHome team, and the maintainence burden also falls to that team. I'm not sure whether that is desirable. What does the team think? 
* A new site a la esphome-devices.com (which is great, by the way) is set up, either under their umbrella or a new organization. I'm willing to look into setting this up if there's interest.
* Any other option? 

Last but not least, of course it'd also need a cool name :)

richardklingler, 2021-05-31:
How about esphome.dev ? (o;

nagyrobi, 2022-07-07:
Yes we need some kind of HACS-like feature for ESPHome...