esphome/feature-requests #1245: PZEM-003T V3

Fionnoch, 2021-06-02:
**Describe the problem you have/What new integration you would like**
The PZEM-003T V3 has a energy monitoring feature which stores the values on the device itself however it can be reset by addressing the address 0x42 using CRC Modbus, an inbuilt feature would be extremely useful as it would allow you to periodically reset the energy monitoring rather than waiting for when the counter reaches its limit and resets itself. 

**Please describe your use case for this integration and alternatives you've tried:**
 An external switch has to made to address this problem and it is not easy to implement as you need to ensure that the correct bytes are sent and that CRC  checkbytes are created

**Additional context**
The command format of the master to reset the slave's energy is (total 4 bytes):
Slave address + 0x42 + CRC check high byte + CRC check low byte.
Correct reply: slave address + 0x42 + CRC check high byte + CRC check low byte.
Error Reply: Slave address + 0xC2 + Abnormal code + CRC check high byte + CRC check
low byte

CRC check use 16bits format, occupy two bytes, the generator polynomial is X16 + X15 +
X2 +1, the polynomial value used for calculation is 0xA001.
The value of the CRC check is a frame data divide all results of checking all the bytes except
the CRC check value

eimertvm, 2021-07-30:
I agree, that would be neat

Guyohms, 2021-09-03:
I'm missing that feature to migrate from Tasmota to ESPHome :)

martgras, 2021-10-05:
If you use the modbus_component instead of the pzem component  this can be done from a template switch 

```yaml
esphome:
  name: pzdc
  platform: ESP32
  board: esp32dev

uart:
  id: mod_bus
  tx_pin: 17
  rx_pin: 16
  baud_rate: 9600
  stop_bits: 2

modbus:
  #flow_control_pin: 23
  send_wait_time: 200ms
  id: mod_bus_pzem

modbus_controller:
  command_throttle: 0ms
  id: pzem017
  ## the Modbus device addr
  address: 0x1
  modbus_id: mod_bus_pzem

sensor:
    - platform: modbus_controller
      id: voltage
      name: "voltage"
      address: 0x0000
      unit_of_measurement: "V" ## for any other unit the value is returned in minutes
      register_type: read
      value_type: U_WORD
      accuracy_decimals: 1
      filters:
        - multiply: 0.01

    - platform: modbus_controller
      id: current
      name: "current"
      address: 0x0001
      unit_of_measurement: "A" ## for any other unit the value is returned in minutes
      register_type: read
      value_type: U_WORD
      accuracy_decimals: 1
      filters:
        - multiply: 0.01

    - platform: modbus_controller
      id: power
      name: "power"
      address: 0x0002
      unit_of_measurement: "W" ## for any other unit the value is returned in minutes
      register_type: read
      value_type: U_DWORD_R
      accuracy_decimals: 1
      filters:
        - multiply: 0.01

    - platform: modbus_controller
      id: energy
      name: "energy"
      address: 0x0004
      unit_of_measurement: "Wh" ## for any other unit the value is returned in minutes
      register_type: read
      value_type: U_DWORD_R
      accuracy_decimals: 0

    - platform: modbus_controller
      id: high_voltage_alarm_threshold
      address: 0x0000
      unit_of_measurement: "V"
      name: "high_voltage_alarm threshold"
      register_type: holding
      value_type: U_WORD
      filters:
        - multiply: 0.01

    - platform: modbus_controller
      id: low_voltage_alarm_threshold
      address: 0x0001
      unit_of_measurement: "V"
      name: "low_voltage_alarm threshold"
      register_type: holding
      value_type: U_WORD
      filters:
        - multiply: 0.01

    - platform: modbus_controller
      id: modbus_rtu_address
      address: 0x0002
      unit_of_measurement: ""
      name: "modbus rtu address"
      register_type: holding
      value_type: U_WORD

    - platform: modbus_controller
      id: current_range
      address: 0x0003
      unit_of_measurement: "A"
      name: "current range"
      register_type: holding
      value_type: U_WORD
      filters:
        - lambda: if (x == 0.0) return 100.0; else if (x == 1.0) return 50.0; else return x * 100.0;

binary_sensor:
    - platform: modbus_controller
      id: high_voltage_alarm
      name: "high voltage alert"
      address: 0x0006
      offset: 12
      register_type: read

    - platform: modbus_controller
      id: low_voltage_alarm
      name: "low voltage alert"
      address: 0x0007
      bitmask: 0xFFFF
      register_type: read

switch:
  - platform: template
    name: "Reset stats"
    optimistic: true
    lambda: |-
       esphome::modbus_controller::ModbusController *controller = id(pzem017);
       std::vector<uint8_t> payload; 
       payload.push_back(0x42); // Add payload - crc will be calculated for you
       esphome::modbus_controller::ModbusCommandItem reset_stat_command =
          esphome::modbus_controller::ModbusCommandItem::create_custom_command(controller, payload);
        // Submit the command to the send queue
        controller->queue_command(reset_stat_command);
        ESP_LOGI("ModbusLambda", "PZEM statistics reset");
        return false ;
```
Notes:
-  the modbus_component  has been merged to the dev branch a few days ago (https://github.com/esphome/esphome/pull/1779)  so you need to use the dev branch for now if you want to test
- broke my pzem 03  last week so I can't test the code above

speakerboar, 2021-10-08:
Thanks @martgras 
for anyone else trying to do the same and get this error
> Too many candidate found for 'modbus_controller_id' type 'modbus_controller::ModbusController' Some are 'pzem1', 'pzem2'.

Add "modbus_controller_id:" to each sensor or  binary sensor

`sensor:
    - platform: modbus_controller
      id: voltage
      name: "voltage"
      modbus_controller_id: pzem1
      address: 0x0000
      unit_of_measurement: "V" ## for any other unit the value is returned in minutes
      register_type: read
      value_type: U_WORD`

Fionnoch, 2021-10-18:
@speakerboar I use 4 sensors at once in my setup if you'd like below is my code you can use to get started should be fairly straightforward to tweak to suit you:
`
uart:
  rx_pin: D4 
  tx_pin: D3 
  baud_rate: 9600
  id: pzem_uart

sensor:
  - platform: pzemac
    address: 0x02
    current:
      name: "Basement Fusebox: Current (Fuse 3)"
    voltage:
      name: "Basement Fusebox:  Voltage (Fuse 3)"
    energy:
      name: "Basement Fusebox: Energy (Fuse 3)"
    power:
      name: "Basement Fusebox: Power (Fuse 3)"
    frequency:
      name: "Basement Fusebox: Frequency (Fuse 3)"
    power_factor:
      name: "Basement Fusebox: Power Factor (Fuse 3)"
    update_interval: 5s
  

  - platform: pzemac
    address: 0x03
    current:
      name: "Shed Current (Fuse 8)"
    voltage:
      name: "Shed Voltage (Fuse 8)"
    energy:
      name: "Shed Energy (Fuse 8)"
    power:
      name: "Shed Power (Fuse 8)"
    frequency:
      name: "Shed Frequency (Fuse 8)"
    power_factor:
      name: "Shed Power Factor (Fuse 8)"
    update_interval: 5s
    
  - platform: pzemac
    address: 0x04
    current:
      name: "Boiler House Current (Fuse 2)"
    voltage:
      name: "Boiler House Voltage (Fuse 2)"
    energy:
      name: "Boiler House Energy (Fuse 2)"
    power:
      name: "Boiler House Power (Fuse 2)"
    frequency:
      name: "Boiler House Frequency (Fuse 2)"
    power_factor:
      name: "Boiler House Power Factor (Fuse 2)"
    update_interval: 5s
    
  - platform: pzemac
    address: 0x05
    current:
      name: "Car Charger Current (Fuse 10)"
    voltage:
      name: "Car Charger Voltage (Fuse 10)"
    energy:
      name: "Car Charger V3 Energy (Fuse 10)"
    power:
      name: "Car Charger Power (Fuse 10)"
    frequency:
      name: "Car Charger Frequency (Fuse 10)"
    power_factor:
      name: "Car Charger Power Factor (Fuse 10)"
    update_interval: 5s
`

You'll have to assign the addresses before being able to use them in esphome. its a quick job with an arduino, just remember the board needs to be powered at the power monitoring side mains for it to work. I've a video where i set up the 4 that might be useful if you're interested :) https://www.youtube.com/watch?v=hao1jUYB5wI&lc=UgwIRYvn-CDXF4WcfVF4AaABAg&ab_channel=FionnOCH

closed 2022-01-10 by jesserockz