esphome/feature-requests #1246: Add transition_length in servo.write

Coninox, 2021-06-03:
**Describe the problem you have/What new integration you would like**

It would be nice if we could set the transition_length parameter in the servo.write action.
We could have the ability to have different rotation speed in different cases.

**Please describe your use case for this integration and alternatives you've tried:**

For example: 
```
api:
  services:
    - service: control
      variables:
        level: float
	transition_lenght: string
      then:
        - servo.write:
            id: my_servo1
            level: !lambda 'return (-2*(level/255))+1;'
	    transition_lenght: !lambda 'return transition_lenght;'

switch:
  - platform: template
    name: boule_ch2
    restore_state: true
    optimistic: true
    
    turn_on_action:
      - servo.write:
          id: my_servo1
          level: -1
          transition_length: "00:00:10"

    turn_off_action:
      - servo.write:
          id: my_servo1
          level: 1
          transition_length: "00:00:40"
```
**Additional context**

narolinus, 2023-04-25:
+1 This would be really great!

narolinus, 2023-04-25:
It is not very intuitive, but I found that you can set the transition length using a lambda call.
` - lambda: |-
            servo_id->set_transition_length(5000);`

where servo_id is the id you set in the servo component definition under `id:`. This workaround is good enough for me for now.

usingapc, 2024-04-24:
@narolinus can you elaborate a little more on your work around?  I'm trying to control both the level and the transition_length of a servo via the API and Input_number helpers from Home Assistant.  I'm able to control the level without issue but can't figure out how to control the transition_length outside of hard coding it in the ESPhome config.  Would you mind sharing more of your config?