esphome/feature-requests #1248: IR remote climate: Add missing Daikin modes

HA-TB303, 2021-06-04:
**Describe the problem you have/What new integration you would like**

The IR Remote Climate daikin component works great, but is missing an ECO, POWERFULL and QUIET mode, als well as the 'night' fan mode. 
[image]

**Please describe your use case for this integration and alternatives you've tried:**

So everytime we go to sleep, we still need to grab the original remote control which kind of defeats the purpose of this great component! This is how we set the system for night, which is not possible using this component:
[image]

**Additional context**

ECO mode can be set with any of the cooling and heating settings and will set the unit to operate 'economically'.
When you set POWERFULL, it will override (all) other settings for 20 minutes and give full cooling/heating power.
QUIET wil just silence the outdoor unit.
The Fan night mode ensures the inside units fan is running in the quit mode.

HA-TB303, 2021-06-04:
If I can do anything to help, like dumping IR-codes, please let me know.

arjen199, 2021-07-13:
I'm also very interested to see support for this being added. Especially the 'fan night mode' option.

jnimmo, 2022-06-14:
Hi all, I'll try and get a PR submitted to fix this in the coming weeks, in the meantime I've added presets in a fork, I just want to sort out the full range of fan speeds before submitting @ https://github.com/jnimmo/esphome/commit/adb7eeb1cd544c4d66801bf64dda38f61b4e63fb

jnimmo, 2022-10-05:
If you're able to provide some feedback on the pull request about this would be good to know how you use the presets (do multiple need to be selected at the same time, or only one at a time?)

closed 2026-10-11 by bdraco