esphome/feature-requests #1257: Add Component Version Control

gb53smith, 2021-06-14:
**Describe the problem you have/What new integration you would like**

Coming from PlatformIO, l like how specific library versions can be specified for the compilation. This allows the user to try another version of a component or even another equivalent component.  For the example the DHT, has many different libraries and versions within those libraries.   I recently fixed an  [issue](https://github.com/esphome/issues/issues/2102) I was having with the DHT using a custom component, in which I specified a DHT library version which worked for me using the PlatformIO IDE.

Reviewing the Esphome code, it looks like component code (.cpp) was copied from somewhere (undocumented) and modified with mainly ESP_LOG functions. Doesn't this make maintenance a nightmare.  How can bug fixes from the original library be passed on the EspHome copy?  Would it not be better to use a wrapper approach like I used for my custom component example?  Bug fixes would then be the responsibility of the ordinating library.  If that failed to fix a bug then a different equivalent library could be substituted, as in the DHT case.

I don't know how easy it would be to move to a wrapper method now?   Maybe is is too late?  It pains me to see issues related to bugs in component internals which should really be fixed in the originating library.  The DHT is a prime example.

If a wrapper was used then a library:  option could be added to platform: to specify the library version.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

OttoWinter, 2021-06-23:
> Reviewing the Esphome code, it looks like component code (.cpp) was copied from somewhere (undocumented) and modified with mainly ESP_LOG functions.

Do you mean you modified the contents of `<CONFIG_FOLDER>/<NODE_NAME>/src/esphome` ? Please see the README in that directory

> THIS DIRECTORY IS AUTO-GENERATED, DO NOT MODIFY
> 
> ESPHome automatically populates the esphome/ directory, and any
> changes to this directory will be removed the next time esphome is
> run.
> 
> For modifying esphome's core files, please use a development esphome install
> or use the custom_components folder

ESPHome works by automatically creating platformio projects for you. You can use the `custom_components` folder (or also `external_components` integration) to modify what code esphome emits.

> Would it not be better to use a wrapper approach like I used for my custom component example?

Sure, and it's already used in a couple of places. But there are some major issues:

- Lots of arduino libraries are very old and not actively maintained (back when we _were_ using this approach I constantly had to fork repos because upstream wasn't responding)
- ESPHome has abstractions for a lot of things, like for example pins. In ESPHome we have `GPIOPin` that can not only represent "normal" GPIO pins, but also port expanders. 

-> In arduino you only have pin numbers, so any library used _would only_ support internal gpio pins. This gets more extreme for example with UART,i2c,SPI where in ESPHome we can use software-based interfaces to enable users to connect to any pins, but Arduino would only support some special pins.

gb53smith, 2021-06-23:
I am aware of the auto generated source code which is copied by a Python program.  I am concerned about a manual copy and paste from a Platformio/Arduino library.  If you can't use the Platformio library and have to "fork" you should add an attribution comment to the components .h file stating:
```This source code originated from <Author>/<Library Name> @ <Version>```
Not version control then but added traceability.  If you find bug in a component you could check with the originating library to see if a new version fixed the bug. 
I switched from the Arduino IDE to Platformio about a year ago. I have found that there is usually another version or even alternate library that I can use to fix an issue I am having.  I have only used 4 component types though.
Special pins are generally fixed for the ESP8266.  For the ESP32, I found this [article](https://randomnerdtutorials.com/esp32-i2c-communication-arduino-ide/) how some libraries support changing I2C pins.

As for ease of use and integration with Home Assistant, ESPhome is a great software innovation.
I also like the ```external_components:``` integration.  It is a better debug method than creating a custom component. I am currently testing different code options for the DHT component.

gb53smith, 2021-06-23:
OK.  Thanks for the explanation and the effort you and your contributors put into this project.
There must be a lot of self-built integrations.  After looking at about 10 components at random, managed to find an attribution in am2320.cpp

closed 2021-06-23 by gb53smith