esphome/feature-requests #1259: "Update" instead of "Install" on the dashboard

noxhirsch, 2021-06-16:
**Describe the problem you have/What new integration you would like**

I understand that it makes sense to use simpler terms, but "install" for an already installed device doesn't make that much sense IMHO (on the device list and in the editor). I'd propose "update" as a simple and well known term, which fits way better.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

mkaatman, 2021-12-30:
I guess it's just personal preferences of semantics but install makes sense to me. You could install a firmware once or many times. You're installing a new firmware every time. If you were actually versioning firmwares and installing version 1.1 over 1.0 then I think "Update" would make more sense to me.

There is "Update All" though which seems to in conflict with "Install"

randybb, 2021-12-30:
To be honest both terms doesn't make sense, in MCU world "flashing" or "load flash" are the right terms

dougle03, 2022-01-22:
For the average user, `update` is more descriptive than either `install` or `flash`.
Whilst all three terms essentially mean the same thing, `update` and `update all` makes more logical sense when _updating_ an existing node IMO.