esphome/feature-requests #1263: Make website more colour blind friendly

kloggy, 2021-06-17:
**Describe the problem you have/What new integration you would like**

Please change the red and green top border indicating on or off line to different colours.

The current colours are extremely colour blind unfriendly.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

wtadler, 2021-06-22:
I'm not colorblind myself, but was just thinking the same. It's subtle even for non-color blind users. Indicators that use both an icon and a color (e.g., ❌ and ✅) would be better.

To illustrate the problem, here's an ESPHome dashboard...
[image]

...and here's that same dashboard run through [this color blindness simulator](https://www.color-blindness.com/coblis-color-blindness-simulator/) for deuteranopia. Good luck quickly determining which devices are connected!
[image]

Red-green colorblindess is very common.

jesserockz, 2021-06-22:
Duplicate of #890

closed 2021-06-22 by jesserockz