esphome/feature-requests #1264: OTA: Automatically build intermediate minimal ROM if binary size exceeds 50% of available flash

MrHaroldA, 2021-06-17:
**Describe the problem you have/What new integration you would like**

I have an ESPHome project which exceeds 50% of the available flash on my ESP8266 based SonOff DUAL R2. It would be nice that instead of an error, ESPHome would build a minimal intermediate ROM and sequentially upload the full ROM automatically. This way I can update all entities if an update shows up ...

**Please describe your use case for this integration and alternatives you've tried:**

I now need to edit my project yaml and comment all, except the OTA settings, build and upload that. After a reboot, I have to uncomment the custom code, sensors, etc and build and upload it again.

Fine-tuning a project is quite labor intensive this way.

**Additional context**

It would be very nice if ESPHome would automatically build the intermediate minimal ROM and upload the full ROM afterwards.

bkaufx, 2021-06-18:
Tasmota allows this type of functionality by providing a "minimal" bin.  I agree it would be cool if the ESPHome dashboard could automatically do this.  Have you tried gzipping the full ESPHome bin?  I'm not sure if the ESPHome dashboard uses gzip but it would be a good temporary solution for you if it's not already being done by the dashboard automatically.

In Linux you can just use the gzip command with the bin file as an argument.  I've done this a few times after downloading the ESPHome bin file if it was too big to flash from a previous Tasmota install so I didn't have to do an intermediate flash to the tasmota-minimal.bin.

vampywiz17, 2021-10-08:
good idea!

Now i try to flash a new firmware (include the new api encrytion) but it say that no enought memory... It need to clear my yaml, flash a simple "only ota" image and after flash the original again... To comlicated, it will be better that we will do it automatically.

nagyrobi, 2022-07-06:
Issue fixed via https://github.com/esphome/esphome/pull/2601
Also see https://github.com/esphome/feature-requests/issues/819

closed 2022-07-06 by nagyrobi