esphome/feature-requests #1266: Nested substitutions

JeeCee1, 2021-06-18:
**Describe the problem you have/What new integration you would like**

Maybe there is already another way to achieve my goal, I just haven't found one.
I have created packages for modules (eg. temperature sensor, pulse counter). For some projects I am using multiple/duplicate sensors, so I have duplicated the packages (e.g. `dht22_1.yaml, dht22_2.yaml` etc.).
I am using substitions for the names, and in the "main" file I am defining these names: 
```
substitutions:
  `dht22_1_name: Elecmeters`. 
```
My request would be to be able to use this:
```
substitutions:
  name_lower: elecmeters
  dht22_1_name: ${name_lower}
```
I know I can directly use `${name_lower}` in the referenced package, but in my situations that is not convenient.
It would suffice to be able to use `#DEFINE` directives directly in the .yaml file, but I guess that conflicts with # being used for comments. 
Is it even possible to use C code directly in the yaml file? 

**Please describe your use case for this integration and alternatives you've tried:**

This would help to mainatain a very clean and lean main project file. 
PS: If anyone knows of a way to avoid having to duplicate the package files that would be great!

**Additional context**

Thanks for any help!

taintedkernel, 2021-06-25:
I have recently stated using packages in my code and realizing I would like the same feature.  It seems quite useful if could create a standard package for a sensor, encapsulating the config and reusing it across multiple instances of the same node.  I have some nodes with multiple instances of the same sensor (ATC MiThermometer), and similar to @JeeCee1 I discovered I cannot leverage packages here.

For a bit more clarification, I would like to be able to do something like the following:

```
# esphome/common/atc_mithermometer.yaml
sensor:
  - platform: atc_mithermometer
    mac_address: ${sensor_mac_address}
    temperature:
      name: "${sensor_name} Temperature"
    humidity:
      name: "${sensor_name} Humidity"
      accuracy_decimals: 1
      filters:
etc...
```

```
# node config esp32-node-1.yaml
substitutions:
  atc1:
    sensor_name: Downstairs
    sensor_mac_address: xx:xx:xx:xx:xx:xx
  atc2:
    sensor_name: Garage
    sensor_mac_address: xx:xx:xx:xx:xx:xx

packages:
  atc_mithermometer: !include common/atc_mithermometer.yaml
```

JeeCee1, 2021-07-25:
ping

joshuaspence, 2021-10-06:
I've also been wanting this, although I imagined it working a different way, similar to the remote package syntax:

```
packages:
  foo:
    path: './foo.yaml`
    substitutions:
      bar: 'baz'
```

joshuaspence, 2021-10-06:
Looks like my suggestion was also made in https://github.com/esphome/feature-requests/issues/805#issuecomment-659669832

nagyrobi, 2022-07-01:
See https://github.com/esphome/esphome/pull/3612