esphome/feature-requests #1270: Digital potentiometers X9C103/X9C104 etc

RomeHein, 2021-06-20:
**Describe the problem you have/What new integration you would like**

X9C103, X9C104 can be found for a dollar or so on aliexpress and can be really useful. 
They are digital potentiometers that can be tricky to use for beginners: https://inventr.io/%F0%9F%8E%9B-how-to-connect-the-digital-potentiometer-x9c102-x9c103-x9c104-to-arduino/
This would be great to have an integration in esphome. Something were all I have to do is set the level of the resistance. At the moment I have to use lambda with for loop to control these.

**Please describe your use case for this integration and alternatives you've tried:**

Atm I use these type of digital potentiometer to control the power of my kitchen hood. But being able to control a digital pot is useful for tons of other things :)

**Additional context**

itakestime, 2021-08-31:
@RomeHein, are you able to share some code on how you go this working in your setup? 

I'm wanting to use a X9C but I see it's not supported out of the box in ESPHome yet.

Strixx76, 2021-10-12:
+1 for supporting X9C-digital potentiometers.

I am about to use two of them for building a digital thermostat for an Nibe Fighter 310p boiler.

Found two Arduino libraries for these:
https://github.com/philbowles/Arduino-X9C
https://github.com/lucyamy/LapX9C10X

Patrik-Hansson, 2022-01-13:
Searching for how to control this component for exactly the same reason. A Nibe Fighter 310p :)

> +1 for supporting X9C-digital potentiometers.
> 
> I am about to use two of them for building a digital thermostat for an Nibe Fighter 310p boiler.
> 
> Found two Arduino libraries for these: https://github.com/philbowles/Arduino-X9C https://github.com/lucyamy/LapX9C10X

MB901, 2022-03-09:
I write some kind of code... but I'm not an expert ! Maybe it will help someone... 
[https://gist.github.com/MB901/8e3991fdd8201f422cf1212073d15f66](url) 
[image]

Patrik-Hansson, 2022-03-10:
Thank you, i had totally given up on this.
Looking forward to trying the code this weekend.

Patrik-Hansson, 2022-08-25:
Sorry for a very late response but this seems to work. Thank you very much.

closed 2023-01-11 by jesserockz

edzima, 2023-01-24:
When can we state this on production?