esphome/feature-requests #1273: Support: Grove - AHT20 I2C Industrial Grade Temperature&Humidity Sensor

sj3fk3, 2021-06-23:
**Describe the problem you have/What new integration you would like**
I would like to use the (Grove)  AHT20 I2C Industrial Grade Temperature&Humidity Sensor with ESPHome

**Please describe your use case for this integration and alternatives you've tried:**
It's a very accurate temp/hum sensor that uses I2C

**Additional context**
To be honest I think the grove ecosystem fits nicely into the ESPHome concept. They have a standardised plug, it works on ESP32's, Pi's and NodeMCU's (M5Stack also uses it).

https://wiki.seeedstudio.com/Grove-AHT20-I2C-Industrial-Grade-Temperature%26Humidity-Sensor/

https://github.com/Seeed-Studio/Seeed_Arduino_AHT20/archive/master.zip

jcastro, 2021-06-23:
Is not working with https://esphome.io/components/sensor/aht10.html ? Just curious as I also want to buy this sensor!

sj3fk3, 2021-06-23:
OMG I am so embarrassed! It works with the aht10 integration. Going to close this one.. No one has seen it right?

closed 2021-06-23 by sj3fk3

jcastro, 2021-06-23:
NO ONE
NO ONE
NO ONE
NO ONE