esphome/feature-requests #1275: PCA9555 16-bit I/O expander support

NdK73, 2021-06-23:
**Describe the problem you have/What new integration you would like**

I'd like to be able to handle PCA9555 like MCP23017.

**Please describe your use case for this integration and alternatives you've tried:**

I tried using MCP23017, but it costs more and does not guarantee output states during boot.

**Additional context**

I'd like to be able to install ESPHome on my DomoNode2 boards. Could a pull-request be useful?

NdK73, 2021-07-27:
Still got no answer, so I wrote a custom device and made it available at https://github.com/NdK73/DomoNode21_for_ESPHome . ESPHome experts can probably simplify it quite a lot.

nagyrobi, 2022-07-07:
Please feel free to submit it as a PR.

closed 2022-07-07 by nagyrobi